espressif/ESP8266_NONOS_SDK #1: SDK is hardly usable with code completion

Dzitu, 2017-04-13:
1. IDEs and code editors provide code completion to increase productivity and convenience,
2. Code completion relies on provided SDK and provided header files. 
3. NON OS SDK should provide verbose header files to enable benefits of code completion.

A lot of functions provided by SDK are not included in header files. For example all functions from os_timer_* family do not have a declaration. Using IntelliSense or other code completion technology is therefore fruitless.

For example:

```
ESP8266_NONOS_SDK> rg ets_timer_
include\osapi.h
48:#define os_timer_arm_us(a, b, c) ets_timer_arm_new(a, b, c, 0)
50:#define os_timer_arm(a, b, c) ets_timer_arm_new(a, b, c, 1)
51:#define os_timer_disarm ets_timer_disarm
52:#define os_timer_setfn ets_timer_setfn

ld\eagle.rom.addr.v6.ld
151:PROVIDE ( ets_timer_arm = 0x40002cc4 );
152:PROVIDE ( ets_timer_disarm = 0x40002d40 );
153:PROVIDE ( ets_timer_done = 0x40002d80 );
154:PROVIDE ( ets_timer_handler_isr = 0x40002da8 );
155:PROVIDE ( ets_timer_init = 0x40002e68 );
156:PROVIDE ( ets_timer_setfn = 0x40002c48 );
```

FayeY, 2017-04-18:
Thanks for your valuable advice. We will optimize it in next SDK.

FayeY, 2017-05-07:
Hi Dzitu, this issue has been optimized in the latest [ESP8266_NONOS_SDK_V2.1](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0), please have a try.

Dzitu, 2017-05-07:
Hello @FayeY, thank you a lot!

closed 2017-05-07 by Dzitu