espressif/ESP8266_NONOS_SDK #2: libgcc.a conflicts with crosstool libgcc

davydnorris, 2017-05-07:
In order for the SDK to be useful, you need a crosstool built toolset of some sort. This will build the latest libgcc for the compiler you choose. The SDK includes libgcc.a as well and this conflicts with the toolchain version.

The only difference between the SDK libgcc.a and the default one built by the toolchain is the removal of several low level math routines that exist in the ROM.

I would recommend that libgcc.a be removed from the SDK, or at least placed somewhere other than with the other libraries. I would also recommend that a script be created that will strip from the library any functions that exist in ROM. This would look something like:

```
export PATH=$(TOOLCHAIN)/bin:${PATH}
cd $(TOOLCHAIN)/lib/gcc/xtensa-lx106-elf/<version>
xtensa-lx106-elf-ar -M < strip_libgcc_funcs.txt

```
[strip_libgcc_funcs.txt](https://github.com/espressif/ESP8266_NONOS_SDK/files/981450/strip_libgcc_funcs.txt)

wujiangang, 2017-06-30:
davydnorris, take some of your suggestions and add to the SDK, thanks!

closed 2017-06-30 by wujiangang

davydnorris, 2017-06-30:
many thanks!