espressif/ESP8266_NONOS_SDK #6: Documentation update for 8MB and 16MB flash support

devsaurus, 2017-05-08:
Thanks a lot for the support of 8MB and 16MB flash chips with v2.1.0! ❤️ 
I've successfully updated the NodeMCU firmware so that 16MB boards like Wemos D1 mini pro are working out of the box (WIP at https://github.com/devsaurus/nodemcu-firmware/tree/align_8mb_16mb).

On the way I figured that the relevant documents are not yet updated to reflect support of 8MB and 16MB. Specifically:
- Chapter 3.3.48 "user_rf_cal_sector_set" in [ESP8266 Non-OS SDK API Reference, 2017.05.05](http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf)
  The case statement is incomplete regarding supported FLASH_MAP_ definitions.
- Chapter 4 "Flash Maps" in [ESP8266 SDK Getting Started Guide, 2017.02.16](http://espressif.com/sites/default/files/documentation/2a-esp8266-sdk_getting_started_guide_en.pdf)
  Flash capacities >4MB are missing.

FayeY, 2017-05-11:
Hi @devsaurus , thanks for your reporting! The documents have been updated, please have a try.

devsaurus, 2017-05-11:
Thanks a lot, again! Your support is very much appreciated 😃

closed 2017-05-11 by devsaurus