espressif/ESP8266_NONOS_SDK #8: esp-now support broadcast?

limin-liu, 2017-05-16:
In your decribetion of code :
WIFI
3. Allow sending a broadcast ESP-NOW packet, but make sure that the packet is unencrypted;

But in your reference V2.1.0 :
3.8 part for ESPNOW  still "ESP-NOW do not support broadcast and multicast.“ 

So ESPNOW actually support broadcast or not？

FayeY, 2017-05-17:
Hi, ESP-NOW does not support broadcast now. 
Where is this description "3. Allow sending a broadcast ESP-NOW packet, but make sure that the packet is unencrypted"?

limin-liu, 2017-05-17:
in the description for this repository's new update code . link here --->https://github.com/espressif/ESP8266_NONOS_SDK/commit/90c641efe84066b47c4616ed367697a9f49f3ac5

Version 2.1.0(116b762)
### System
1. Fix and update the license;
2. Add definitions of ROM functions;
3. Add efuse mac crc, save efuse mac into the system parameter area of flash, as a backup mac;
4. Implement `uart_div_modify`, which can support a 32-bit DivLatchValue in the SDK instead of the ROM one, and also remove `uart_div_modify` from the ld file;
5. Reduce iRAM usage by transferring some functions to flash；
    - add weak function `void user_spi_flash_dio_to_qio_pre_init(void)`. If the dio to qio flash mode is not used, users can add an empty `user_spi_flash_dio_to_qio_pre_init` on the application side to reduce iRAM usage;
6. Fix WDT in system_restart;
7. Fix Write & Read issue in SPI overlap mode;
8. Support 8MB & 16MB spi flash;
9. Update boot v1.7;
    - Support 8MB & 16MB spi flash;
    - Try boot the first bin for 3 times. If it fails, then boot the backup bin;
10. Fix http packet error, BBP301;
11. Support long periods of deep sleep;
12. Fix NMI handle crash on unaligned memory access, BBP309;

### Lwip
1. Add set and get dns server APIs:
    - espconn_dns_setserver
    - espconn_dns_getserver
2. Fix 2038 overflow bug in sntp_get_current_timestamp;
3. Fix MTU negotiate bug;

### AT
1. Add new commands, AT+UART? and AT+UART_DEF?;
2. Add new command, AT+SYSRAM?;
3. Add new commands, AT+SYSIO series;
4. Add new command, AT+CWHOSTNAME;
5. Add new command, AT+SYSADC?;
6. Add new commands, AT+CIPSNTPCFG and AT+CIPSNTPTIME;
7. Add new commands, AT+CIPDNS_CUR and AT+CIPDNS_DEF;
8. Support a maximum of 8 stations;
9. Modify keepalive count to 3 times;
10. Fix scan hidden ssid issue in AT+CWLAP;
11. Fix the issue of the UART stop bit being 0 when queried for the first time;
12. Fix MAC addr conflict in AP+STA mode;
13. Add new commands, AT+RFAUTOTRACE;

### WiFi
1. Optimize the RF frequency calibration workflow, having the frequency trace disabled by default;
    - users can call `system_phy_freq_trace_enable` in `user_rf_pre_init` to enable it;
2. Update libphy.a to version 1134_0;
3. Allow sending a broadcast ESP-NOW packet, but make sure that the packet is unencrypted;
4. Fix soft queue error;
5. Fix EVENT_STAMODE_AUTHMODE_CHANGE event mode info;
6. Fix lmac assert;
7. Fix bug in sniffer mode;
8. Fix arp loss continously issue;
9. Fix bug in WiFi event;
    - auth mode changed event cannot be throw when station is in the process of scanning;
    - error old auth mode and new auth mode;
    - op mode change process cannot be blocked, so add an event to notify the caller that op mode is changed;
10. Fix the bug in mic failure handler;
11. Fix the bug which causes failure of connection to an AP with shared WEP;
12. Fix the bug which causes failure of connection to an AP with hidden ssid;
13. Fix softAP deauth;
14. Fix reconnect policy when trying to connect to an AP that does not exist;
15. Fix scan issue;
16. Fix extra bytes in UDP broadcast packet, BBP310;

FayeY, 2017-05-17:
Hi, thanks for your reporting. The documentation has been updated, see [API Guide](http://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf).
ESP-NOW supports broadcast in V2.1.

franklinvv, 2017-08-08:
How can we perform a broadcast with ESP-NOW? Do we transmit to mac address {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF}? Because I tried that and it didn't seem to work ...

Daniel-dk, 2017-11-27:
from the header file it seems we have to set it to "null"
"
```
/**
  * @brief     Send ESPNOW data
  *
  * @attention 1. If peer_addr is not NULL, send data to the peer whose MAC address matches peer_addr
  * @attention 2. If peer_addr is NULL, send data to all of the peers that are added to the peer list
  * @attention 3. The maximum length of data must be less than ESP_NOW_MAX_DATA_LEN
  * @attention 4. The buffer pointed to by data argument does not need to be valid after esp_now_send returns
  *
  * @param     peer_addr  peer MAC address
  * @param     data  data to send
  * @param     len  length of data
  *
  * @return
  *          - ESP_OK : succeed
  *          - ESP_ERR_ESPNOW_NOT_INIT : ESPNOW is not initialized
  *          - ESP_ERR_ESPNOW_ARG : invalid argument
  *          - ESP_ERR_ESPNOW_INTERNAL : internal error
  *          - ESP_ERR_ESPNOW_NO_MEM : out of memory
  *          - ESP_ERR_ESPNOW_NOT_FOUND : peer is not found
  *          - ESP_ERR_ESPNOW_IF : current WiFi interface doesn't match that of peer
  */
esp_err_t esp_now_send(const uint8_t *peer_addr, const uint8_t *data, size_t len);
```
I don't have multiple ESP32's to test with yet, will report once I do

franklinvv, 2017-11-27:
That’s not a real broadcast if you ask me. I haven’t looked at the source code, but I presume that it just loops through the peers one by one, doesn’t it?

It would be cool to allow ESPs to discover each other by broadcasting, and then adding each other as peers.

Daniel-dk, 2017-11-28:
It seems its a real broadcast ..
if you send unencrypted data to peer addr ff:ff:ff:ff:ff:ff then any device on the same channel will receive it
You need to add a peer with Addr ff:ff:ff:ff:ff:ff to the peer list otherwise it doesnt send the packet..

Device A :

```
Sending: 29
Send Status: Success
Last Packet Sent to: ff:ff:ff:ff:ff:ff
Last Packet Send Status: Delivery Success
```

```
Device B  : 
Last Packet Recv from: 24:0a:c4:yy:yy:yy
Last Packet Recv Data: 29
```

test .ino here ( quick modified version of the esp-Now example, flashed to 2 ESP32 dev boards )
[Gist link](https://gist.github.com/Daniel-dk/0a31ca66240e076f270d2d65bb4ea7f1)

franklinvv, 2017-11-30:
Oh, then I stand corrected!

However, I only have one ESP32 so I can't test it on the ESP32. I have multiple ESP8266s so I will have a try asap. :)

Crazor, 2018-05-29:
I encountered the same issue as @cnlohr where I can successfully send to FF:FF:FF:FF:FF:FF, but those broadcast packets are never received.
@Daniel-dk said it is needed to add the broadcast address as a peer, however that fails with a return code of -3 (I haven't found any explanation of this function's return codes).
I'm on SDK 2.2.1 (via the Arduino core 2.4.1)

cnlohr, 2018-05-31:
@FayeY Can we please re-open this issue as it is definitely present in the current nonos sdk.  I am stuck back at 1.5.2 for projects where this is needed.

littleyoda, 2018-06-13:
I have the same Problem with esp8266. I can send boardcasts, but no esp8266 receive them.

Crypter, 2018-08-22:
I can confirm this problem too. Devs, do we have anyone working on this?

Thanks

cnlohr, 2018-08-27:
I'm not really sure how to get their attention...

Crypter, 2018-08-27:
@cnlohr, as far as I'm concerned they might as well have abandoned esp8266 and are working on esp32. Most of them are dedicated to direct customers of espressif (for example alibaba), with features going into the SDK only if they need them for the customers.

If only everything could be open source... 🙄

kriegste, 2018-08-27:
Try emailing bugbounty@espressif.com

FayeY, 2018-08-28:
@Junhao-Espressif, PTAL

cnlohr, 2018-08-31:
@FayeY I am glad that Espressif is pursuing this again.  It is much more powerful than any framework could be.  By having true broadcast access, it opens a lot of doors that are unworkable with different options.

cnlohr, 2018-09-08:
@FayeY I am coming up on a commercial project that will be using 2,500 units and I really could use this feature.  I don't want to stick the customer with the old firmware version.

Junhao-Espressif, 2018-09-10:
Hi @limin-liu @franklinvv @Daniel-dk @cnlohr 

Please try using the following two files. After testing, we will update them on GitHub.

[lib_support_broadcast.tar.gz](https://github.com/espressif/ESP8266_NONOS_SDK/files/2365002/lib_support_broadcast.tar.gz)

```
SDK ver: 3.0.0(e27cf60) compiled @ Sep  5 2018 15:35:05
phy ver: 1136_0, pp ver: 10.2

SDK version:3.0.0(e27cf60)
mode : sta(5c:cf:7f:14:c7:8a)
add if0
esp_now init ok
==================
CONTROLLER
==================
FF FF FF FF FF FF SEND SUCCESSFUL!
now from[5E CF 7F 14 C6 4A  len: 13]:ACK[00000000]
FF FF FF FF FF FF SEND SUCCESSFUL!
now from[5E CF 7F 14 C6 4A  len: 13]:ACK[00000001]
FF FF FF FF FF FF SEND SUCCESSFUL!
now from[5E CF 7F 14 C6 4A  len: 13]:ACK[00000002]
FF FF FF FF FF FF SEND SUCCESSFUL!
now from[5E CF 7F 14 C6 4A  len: 13]:ACK[00000003]
FF FF FF FF FF FF SEND SUCCESSFUL!
FF FF FF FF FF FF SEND SUCCESSFUL!
now from[5E CF 7F 14 C6 4A  len: 13]:ACK[00000004]
FF FF FF FF FF FF SEND SUCCESSFUL!
```

```
SDK version:3.0.0(e27cf60)
mode : softAP(5e:cf:7f:14:c6:4a)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
esp_now init ok
==================
SLAVE
==================
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!
5C CF 7F 14 C7 8A SEND SUCCESSFUL!
recv action mgment
now from[5C CF 7F 14 C7 8A  len: 12]:Hello World!

```

cnlohr, 2018-09-10:
I cannot look into this for another few days.  But, I am eagerly going to try it out when I can!

AEFeinstein, 2018-10-27:
@Junhao-Espressif I gave the libs posted on September 9th a try, and they worked! In the setup I called esp_now_add_peer() with NULL for the key, and 0 for the key length and was able to broadcast and receive unencrypted packets. Thanks!

AEFeinstein, 2018-10-27:
Also I did my testing with @cnlohr, so I think those libs can be uploaded to the repository.

torntrousers, 2018-11-04:
Is there a way to try the fix in lib_support_broadcast.tar.gz with the Arduino/ESP8266 code? When I try updating with the two files in lib_support_broadcast.tar.gz then all the compiles fail with: 
```
C:\Arduino\ESP8266\181104\arduino-1.8.7\hardware\esp8266com\esp8266/tools/sdk/lib\libpp.a(pm_for_bcn_only_mode.o): In function `fpm_open':

/home/jhzhang/workspace/source_code/ESP8266_NONOS/esp8266_nonos_sdk_core/app/pp/pm_for_bcn_only_mode.c:227: undefined reference to `system_rf_cal_sector'

C:\Arduino\ESP8266\181104\arduino-1.8.7\hardware\esp8266com\esp8266/tools/sdk/lib\libpp.a(trc.o): In function `rcBuildIndex':

/home/jhzhang/workspace/source_code/ESP8266_NONOS/esp8266_nonos_sdk_core/app/pp/trc.c:414: undefined reference to `open_signaling_measurement'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
``` 
Thats with the latest (today) Arduino/esp8266 github code where system_get_sdk_version() returns: 3.0.0-dev(c0f7b44)

AEFeinstein, 2018-11-09:
I see libespnow.a and libpp.a were modified in b7ac6686ff7b906f6c80a45df02956e94d4b4566, but they are not identical to the files from lib_support_broadcast.tar.gz. @xcguang are the only differences compilation options, or did something else change?

DavidAC95, 2018-12-13:
@Junhao-Espressif Can you tell me how to implement this fix for a normal Esp8266 with Arduino IDE? As I cannot figure out how to install that file you posted, and I really need to be able to broadcast to get my project working. @AEFeinstein Or can you tell me how you got it working, if you were using an Arduino IDE that is?

AEFeinstein, 2018-12-13:
I was not using the Arduino IDE, I was using the non os sdk directly with a makefile on Linux. Sorry I cannot help you.

Junhao-Espressif, 2018-12-18:
Hi @DavidAC95 ,

Could you tell me the SDK verison ?

torntrousers, 2018-12-18:
@Junhao-Espressif the most recent ESP8266 core for the Arduino IDE (2.5.0-beta2) was released just yesterday and that uses SDK version 3.0.0-dev(c0f7b44).

marcelstoer, 2018-12-18:
@torntrousers 
> that uses SDK version 3.0.0-dev(c0f7b44)

not according to this file: https://github.com/esp8266/Arduino/blob/2.5.0-beta2/tools/sdk/version#L1

torntrousers, 2018-12-18:
Mmm, interesting. I just now did a ```Serial.println(system_get_sdk_version()); ``` on an ESP running the latest release and 3.0.0-dev(c0f7b44) is what that outputs.

DavidAC95, 2018-12-18:
@Junhao-Espressif I dont have the system in front of me at this moment. And I won't for a few weeks now but the last library update I did was for esp8266 2.4.2 which was on Dec 13th.

canadaduane, 2018-12-31:
I too am interested in the Arduino IDE release of this fix.

DavidAC95, 2019-01-03:
@Junhao-Expressif has any progress been made on this issue for the Arduino IDE?

d-a-v, 2019-01-03:
@ arduino users, please understand that espressif cannot answer here for arduino.
Check the referenced issue above directing to the esp8266/arduino repository.
Short answer, arduino still has not nonos-sdk-v3. Upgrading to this version is not straightforward as it was with older versions and we are currently busy with arduino core 2.5.0 stabilization.
With espressif's vision of nonos-sdk's future, we need to think a bit more before our next step.

DavidAC95, 2019-01-03:
@d-a-v I understand, then we will probably switch to the esp32 as we need broadcast capability. I do have one more question though, can the esp32 talk to esp8266? As in if we have the main hub being a esp32 could it broadcast to the esp8266s?

leenowell, 2019-01-23:
Hi

I have this issue too and have tried 2.5 beta2 and 2.4.2 and neither seem to work.  I have changed the 2 libraries from 2.5 beta 2 to the ones about and then get linker errors.  Anyone have broadcasting working?

@Junhao-Espressif any update on the bug?

@DavidAC95 I have an esp32 sending an espnow message to an esp8266 (directly using its mac address) and the esp8266 receiving it.  Broadcasting from the esp32 to the esp8266 is what doesn't seem to work.

thanks

Lee.

theicfire, 2019-08-19:
Here's the fix in the Arduino library: https://github.com/esp8266/Arduino/issues/6174

petrkucerak, 2023-04-04:
In my experience, the Arduino framework uses a library that implements only pseudo broadcast.[[1](https://github.com/yoursunny/WifiEspNow)] So if you want to simply send messages to all devices, you will be satisfied. Protocol will not function if you have any requirements for time.

You may want to use the ESP-IDF framework based on freeRTOS. This version supports broadcast too. It isn't a real broadcast also.

But I'll be planning to inspect this problem in more detail. I'll actualize the information.