espressif/ESP8266_NONOS_SDK #10: Bug in espconn_secure_send

davydnorris, 2017-05-24:
See:
[http://bbs.espressif.com/viewtopic.php?f=66&t=4649](url)
and
[http://bbs.espressif.com/viewtopic.php?f=16&t=3423](url)

espconn_secure_send() appears to have a bug that means it will only correctly send once per secure connection. The second time you call it, no data is sent and it returns error -5 indicating there's an operation in progress.

This is after the previous send fires the sent callback function showing a successful send, and the espconn state for the connection shows that it's ready to send again.

This bug breaks any program where multiple secure sends are required on the same connection, such as serving web pages, or sending large amounts of data.

FayeY, 2017-05-28:
Hi @davydnorris , can you provide your test code for debugging?

davydnorris, 2017-05-28:
Sure - this is code directly from the ESP-HTTPD project by Sprite_tm, with modifications in core/http-nonos.c to support SSL.

I have replaced my certificate information with the files shipped in the IoT_Demo - you will need to adjust the web site name in user_main.c to match the name in the SSL certificate you use.

Compile with ESP-NONOS SDK 2.1.0

[esphttpd_espressif.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1033936/esphttpd_espressif.zip)

FayeY, 2017-05-31:
Hi @davydnorris , yes, this is a known issue, would you mind to use [mbedTLS](http://www.espressif.com/en/support/download/sdks-demos) instead?

davydnorris, 2017-06-02:
Hi @FayeY,

Tried mbedTLS last night (simply switched libraries and nothing else), and found the following:
 - handshake still failing on smartphones connected through ESP AP (self signed Certificate is registered on phone)
 - handshake succeeds from browser connecting through regular network where ESP is connected as STA
 - first page and all content transferred successfully on first secure connection
 - handshake fails for every connection afterwards. No further data can be transferred

Could you please use the same code, same Makefile and the mbedTLS library to check this out and see if you see the same behaviour?

valkuc, 2017-06-12:
Same here. Related BBS topic http://bbs.espressif.com/viewtopic.php?f=7&t=1791
Reproducing on SDK 2.1.0. Figured out that first connection work fine. Any next calls produce Fatal Exception. Here is part of log:
```
DEBUG: Free heap size: 31536 bytes. (mqtt_service.c:927)
INFO: MQTT service started.
client handshake start.
E:M 2056
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

```

valkuc, 2017-06-21:
Any news on this issue?

donghengqaz, 2017-06-22:
I see “E:M 2056”, it means no heap memory can be supplied...

valkuc, 2017-06-22:
Please pay attention on line before:
`DEBUG: Free heap size: 31536 bytes`
It was issued just before new esp conn was called. You want to tell that secure connection try to allocate 29Kb of heap?!! What is located at address 0x4000e1b2?

valkuc, 2017-07-07:
Well, I have performed clean analisys and found that issue actually is not related to espconn_secure_* methods. It's actually related to dynamic heap memory allocation for buffers used in MQTT (offline message queue buffer) and it's fragmentation. Decreasing queue buffer size solved the problem. The most intersting thing is that required memory for my needs was about 6 Kb while I have 32 Kb free memory. So as I said before I guess that the problem is in memory fragmentation.

davydnorris, 2017-07-08:
I get these issues purely with esp_secure* routines - I'm not using MQTT, I'm using the HTTPServer project modified to provide HTTPS

valkuc, 2017-07-08:
In general this is not related to MQTT, it's related to memory allocation. In my case memory was allocated in MQTT module, in your case - it's probably in HTTPServer project. I just wrote a simple test case:
1) open secure connection to host
2) send some bytes
3) disconnect from host (with full cleaning of connection, i.e. espconn_abort, espconn_secure_delete)
4) start again from p.1 for infinite time.
I have run this test for about few hours and it work like a charm without any problems.

davydnorris, 2017-07-12:
@valkuc - which libraries are you using in your code? I find the following with SDK2.1.0:
 - with libssl I get one call per connection to esp_secure_send that works then next calls all fail. So for instance with EspHttpd clients will connect securely, then request a page and get the 200 header back OK, but then content will not send on that same connection.
 - with libmbedtls and the scenario above I get one complete connection worth of traffic back and forth but then any other connections fail handshake.

Can you show me your disconnect code please? I am pretty sure I am cleaning up completely but I don't use abort or delete while the server is running - I am using secure_disconnect only, and secure_delete only when I stop the listener.

valkuc, 2017-07-12:
I'm using https://github.com/Spritetm/libesphttpd but without SSL. So, SSL in my project is used only for connecting to secure server in MQTT project ( https://github.com/tuanpmt/esp_mqtt ). Disconnect code you can find in this MQTT lib.

valkuc, 2017-12-28:
Bump... Espressif, any progress on this issue?

FayeY, 2017-12-29:
Hi, sorry that SSL do need a lot of memory. mbedTLS is a third party libaries, we have already optimized its memory occupation. But it still needs at least 22KB of memory size to run the SSL function. And if you use a large certificate on your SSL server, the memory it needed will be even more.  
Sorry for the inconvenience.

valkuc, 2017-12-29:
But what if I'm using very small certificate? Say, 512 bit... or 1024 bit?

FayeY, 2018-09-20:
We have provided a way to enlarge memory in NonOS SDK V3.0. Please have a try. Thanks.

closed 2018-09-20 by FayeY