espressif/ESP8266_NONOS_SDK #16: Include math lib

kriegste, 2017-06-21:
Hi,

please include the math lib in the NONOS SDK so we can use sin, cos, asin, atan2 and the like.

Please compile this and all the other libs with parameters -ffunction-sections -fdata-sections which will open a chance to shrink the images down quite a bit. 512kB are often a bit tight.

Thank you very much for your great work!

wujiangang, 2017-06-29:
hi, @kriegste 

If you want to use math functions, you can add -lm in your Makefile, and include math.h in your source code.

kriegste, 2017-06-29:
The problem with the standard -lm is...

> /opt/Espressif/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/httpd.user1.out section `.text' will not fit in region `iram1_0_seg'

I think in the RTOS SDK you included an irom version of lib m. My request would be to include this in the NONOS SDK as well.

wujiangang, 2017-06-29:
OK, got your point, will update soon.

wujiangang, 2017-06-29:
Please test commit: 33d4ee2

kriegste, 2017-06-29:
Compilation works and calculations seem to be correct, thank you very much!

wujiangang, 2017-06-30:
Great, compile libs with parameters -ffunction-sections -fdata-sections will be committed soon. This is a topic of #13 .

closed 2017-06-30 by wujiangang