espressif/ESP8266_NONOS_SDK #18: SDK 2.1.0 - Boot sequence Issue

yrvverma, 2017-06-26:
I tried using SDK 2.1.0 for flashing. But this seems to have some issues during boot sequence. Please find SDK 2.0.0 output after successful flash.  Also attached error output after SDK 2.1.0 successful flash. 

[output-boot-1.6bin.txt](https://github.com/espressif/ESP8266_NONOS_SDK/files/1101203/output-boot-1.6bin.txt)
[output-boot-error1.7bin.txt](https://github.com/espressif/ESP8266_NONOS_SDK/files/1101204/output-boot-error1.7bin.txt)

valkuc, 2017-06-26:
Using SDK 2.1.0 and boot 1.7 for more than 2 weeks - haven't found any issues with new boot.

yrvverma, 2017-06-26:
ok  will try again and revert. One issue I did find in my analysis was that I was using 0x7C000 address for esp_init_data_default. Let me change to 0xfc000 and re-flash with SDK 2.1.0. Will update on results.

kriegste, 2017-07-02:
All is fine here, too.

The only problem I experienced was when I once used system_restart_enhance() it screwed up my OTA firmware update function completely: I had the idea of using system_restart_enhance() to "try" a newly written firmware image first. The idea was that if it crashes, I can simply poweroff/poweron the module to revert back to the old image.
But that did not work. The new boot address got stored somewhere and the module booted the new firmware every time (system_get_userbin_addr showed the new address). system_upgrade_userbin_check instead said the old image was active.

So in fact
system_upgrade_userbin_check reported 1
and at the same time
system_get_userbin_addr reported 0x001000

I had to reflash the module. But since I stopped my attempts with system_restart_enhance all is fine again.