espressif/ESP8266_NONOS_SDK #20: User proper commit messages on GitHub

marcelstoer, 2017-06-29:
I believe this says it all..

[image]

It's not practical to open each commit to figure out what the changes actually were. The first line, ideally no more than 50 chars, should contain the essence of the commit.

wujiangang, 2017-06-30:
Sometimes, the commit header cannot describe all modifications, since we will sync several changes from our internal gitlab, so just take this as a merge. If you want to know the changes well, use `git log` will show more info.

marcelstoer, 2017-06-30:
> since we will sync several changes from our internal gitlab

I suspected that this is the reason but it doesn't _have_ to be that way. GitLab can be told to [automatically push changes to a remote repo](https://docs.gitlab.com/ee/workflow/repository_mirroring.html#pushing-to-a-remote-repository) such as GitHub.

> If you want to know the changes well, use `git log`

I know, but for casual observers (what most non-SDK developers are) this isn't practical because you'd have to clone the repository first.

You write [useful release notes](https://github.com/espressif/ESP8266_NONOS_SDK/releases) for people who are interested in the changes between release N and release N+1. Thanks for that. If I want to figure out what the changes since the latest release are I would look at the commit history as shown above.

wujiangang, 2017-06-30:
The github repo is not a mirror of gitlab repo, so cannot use the way you provide. Now, it's some inconvenient for you, i will modify the commit way, thanks for your advise.

closed 2018-09-20 by FayeY