espressif/ESP8266_NONOS_SDK #22: USE_OPTIMIZE_PRINTF for os_sprintf

kriegste, 2017-07-01:
There is USE_OPTIMIZE_PRINTF which moves strings automatically from RAM to ROM when using

os_printf

Now, it would be great if this also moved the strings in

os_sprintf

from RAM to ROM.
Building responses in a tcp server often needs a lot of data being formatted, so there are a lot of strings involved. It is bad if they end up in RAM.

kriegste, 2017-12-05:
```
#define os_sprintf(str, fmt, ...) (__extension__({ \
	static const char FMT[] ICACHE_RODATA_ATTR STORE_ATTR = fmt; \
	int cccccccc = 0; \
	cccccccc = os_sprintf_plus(str, FMT, ##__VA_ARGS__); \
	cccccccc; \
}))

int ICACHE_FLASH_ATTR os_sprintf_plus(char *str, const char *format, ...) {
	int flen = (strlen(format) + 4) & ~3;
	char *f = (char *)os_malloc(flen);
	os_memcpy(f, format, flen);
	va_list args;
	va_start(args, format);
	int len = ets_vsprintf(str, f, args);
	va_end(args);	
	os_free(f);
	return len;
}
```

something like that...

kriegste, 2018-07-13:
I have seen attempts implementing the feature.
https://github.com/espressif/ESP8266_NONOS_SDK/commit/cac83b12994f5dd06ee4dc168dc0dfb892c7fb46#diff-3282dd557c8bea613997e920bd301c1f
There are two problems:
First, if someone uses a variable "len" in the same scope, the program will crash. If should be renamed to "__len__" or something other unlikely used name.
Secondly, there seems to be a problem with the function when multiple arguments are used. I need to investigate further. My implementation from December (see the last post) is ok.

kriegste, 2018-07-13:
```
	int len = os_sprintf(buff, "\"SDK\":\"%s\", \"Mode\":%d, \"Freq\":%d, \"Heap\":%d, \"Phy\":%d, \"Sleep\":%d, \"Map\":%d, \"Boot\":%d, \"Bin\":%d, \"Addr\":\"0x%05X\", \"FlashID\":\"0x%06X\"",
		system_get_sdk_version(), wifi_get_opmode(), system_get_cpu_freq(), system_get_free_heap_size(), wifi_get_phy_mode(), wifi_get_sleep_type(), system_get_flash_size_map(),
		system_get_boot_version(), system_upgrade_userbin_check(), system_get_userbin_addr(), spi_flash_get_id());
```
This is existing code that worked with my December solution and now crashes with a watchdog exception. The line isn't that long, is it? It also demonstrates the problem with the "len" variable.

xcguang, 2018-07-14:
Hi @kriegste ,
Today I found os_sprintf will crash in some case. And I'm debugging, but there's no result yet. If it blocks your developing, please redefine os_sprintf with os_sprintf_plus, very sorry for the inconvenience.

kriegste, 2018-07-14:
No problem. I returned to the old osapi.h file and my own solution from above. It works great. Use it if you like.

kriegste, 2018-10-01:
What is the current status of this issue? I am not sure, did you revert the optimization you attempted or does the current SDK use a fixed version?

MaBecker, 2019-06-06:
> ```
> #define os_sprintf(str, fmt, ...) (__extension__({ \
> 	static const char FMT[] ICACHE_RODATA_ATTR STORE_ATTR = fmt; \
> 	int cccccccc = 0; \
> 	cccccccc = os_sprintf_plus(str, FMT, ##__VA_ARGS__); \ [...]

hi, is this still the version you use for your project?

kriegste, 2019-06-06:
Yes. The define is in an .h file. The function is in a .c file. This is the prototype (.h file):
int os_sprintf_plus(char *str, const char *format, ...) __attribute__ ((format (printf, 2, 3)));

I undef the existing os_sprints beforehand. No problems since I posted it.

MaBecker, 2019-06-06:
Thanks for your quick reply, will  implement it for  Espruino ;-)