espressif/ESP8266_NONOS_SDK #35: Issue with CIPSTART and wrong link ID

exeron, 2017-08-18:
I noticed that you can connect to a server using wrong channels. See the next results from the ESP to understand (domain removed obviusly for privacy):
With ESP8266 in STA-AP mode and correctly joined to a WiFi spot:

AT+CIPSTART=**-10**,"TCP","xxxxxxxxxxx",30000
**246,CONNECT**
OK

And now I was able to send data to the ESP without problems.

**+IPD,246,6:aaaa
+IPD,246,12:asdadasdad**

With others IDs i got some error messages but not very related:

AT+CIPSTART=-3,"TCP","xxxxxxxxxxxxx",30000
ALREADY CONNECTED
AT+CIPSTART=-4,"TCP","xxxxxxxxxxxxx",30000
DNS Fail

Some blocks of negative numbers returns "ALREADY CONNECTED" error.
Other negative numbers (the most) returns "DNS Fail".
With positive IDs above 4 the message is "ID ERROR" what is expected since channels are limited to 0-4. 
And as you can see.. at least id -10 is allowed and connected perfectly.

PS:
AT version:1.4.0.0(May  5 2017 16:10:59)
SDK version:2.1.0(116b762)
compile time:May  5 2017 16:37:48

I wrote in the forum but looks people is not interested in answering anything...

FayeY, 2017-08-23:
Why do you need a negative number? The doc has already said the range should be [0, 4]. Sorry for the inconvenience.

FayeY, 2017-08-23:
Hi,  the range should be [0, 4], but anyway, we will revise it in the next AT firmware. Sorry for the inconvenience.

exeron, 2017-08-24:
I know the range is 0-4.  It's not an inconvenience. An inconvenience is not being able to limit the number of connections of the ESP when working in **STA mode as a Server**, that is how I noticed this problem with link ids.

The point is some link ids out of range are allowed and **working**, therefore the range is **not** 0-4 -apparently-. And error reporting is also different depending on the numbers what can make you think they are due to different reasons. I also noticed after entering some wrong link ids the ESP became erratic and eventually it can reseted without further notice.

The channels I noticed working with the wrong id, suggest a signed integer is being assigned to an unsigned integer and the 2's complement does the rest (I cannot go further without having the firmware); and about the "irregular" messages surely where the range is checked some branch is not expected and that's the reason to the not related messages, or apparently not related.

FayeY, 2018-04-16:
Solved in the latest SDK.

closed 2018-04-16 by FayeY