espressif/ESP8266_NONOS_SDK #37: mbedTls NONOS SDK 2.1 HTTP Server fatal exception

crgor, 2017-08-23:
Hi All,

I have been playing around with new libmbedtls.a library downloaded from espressif SDK (precompiled from here [http://espressif.com/sites/default/ ... 160718.zip]) .

It seems there is an issue running IOT Demo HTTP server SSL enabled using mbedTls library. Once I start a client connection this is what happens

> === System init ====
> mode : softAP(62:01:94:84:0d:2e)
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> bcn 100
> add 1
> aid 1
> station: 90:61:0c:32:59:29 join, AID = 1
> server handshake start.
> server handshake ok!
> server's data invalid protocol
> Reason:[-0x7880]
> webserver's 192.168.4.33:13907 err 8 reconnect
> server handshake start.
> server handshake ok!
> Fatal exception 28(LoadProhibitedCause):
> ÿpc1=0x40001b7f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
> ets Jan 8 2013,rst cause:1, boot mode:(3,7)
> 
> load 0x40100000, len 2592, room 16
> tail 0
> chksum 0xf3
> load 0x3ffe8000, len 764, room 8
> tail 4
> chksum 0x92
> load 0x3ffe82fc, len 676, room 4
> tail 0
> chksum 0x22
> csum 0x22
> 
> 2nd boot version : 1.7(5d6f877)
> SPI Speed : 40MHz
> SPI Mode : DOUT
> SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 

I have set the SSL settings as follows on the HTTP server
```
espconn_secure_set_size(ESPCONN_SERVER, 4096);
         espconn_secure_set_default_certificate(default_certificate, default_certificate_len);
        espconn_secure_set_default_private_key(default_private_key, default_private_key_len);
        espconn_secure_accept(&esp_conn);
```

I have tested by increasing the size of espconn (6*1024), same happens.

Any idea, why it's happening? What am I missing?

marcelstoer, 2017-08-23:
Is this related to #10 ?

valkuc, 2017-08-23:
Yes, it can be, but who knows? Espressif are not very active here :(

FayeY, 2017-08-23:
Hi @crgor , can you provide your test code for debugging?

FayeY, 2017-08-23:
The IOT Demo is not using the mbedTLS, you need to modify the makefile to use mbedTLS instead. We have tested it with mbedTLS, it works fine.

crgor, 2017-08-24:
@FayeY 
Yes, I have changed the make file to use mbedtls instead of SSL and I have enabled IOT Demo code to start the web server. It does work fine with out this extra piece (Basic realm authentication) which I added in the web server, it looks as follows

`
/******************************************************************************
 * FunctionName : default_basic_authentication_callback
 * Description  : check for HTTP basic authorization credentials
 * Parameters   : user
 *                passwd
 * Returns      : boolean
 *******************************************************************************/
LOCAL bool ICACHE_FLASH_ATTR default_basic_authentication_callback(char *user,
		char *passwd) {
	//os_printf("Authentication %s %s %s %s\n", user, passwd, sysCfg.httpd_user, sysCfg.httpd_pass);
	if (os_strcmp(user, sysCfg.httpd_user) != 0) {
		return false;
	}

	if (os_strcmp(passwd, sysCfg.httpd_pass) != 0) {
		return false;
	}

	return true;
}

/******************************************************************************
 * FunctionName : webserver_authentication
 * Description  : check for HTTP authorization credentials
 * Parameters   : user
 *                passwd
 * Returns      : boolean
 *******************************************************************************/
bool ICACHE_FLASH_ATTR webserver_authentication(char *user, char *passwd) {
	return default_basic_authentication_callback(user, passwd);
}

/******************************************************************************
 * FunctionName : strstr_end
 * Description  : finds the first occurrence of the substring needle in the string haystack
 * Parameters   : haystack
 *                needle
 * Returns      : pointer to the end of the first occurrence
 *******************************************************************************/
void ICACHE_FLASH_ATTR *strstr_end(char *haystack, char *needle) {
	char *pStart;
	pStart = (char *) os_strstr(haystack, needle);
	if (pStart == NULL) {
		return NULL;
	}

	return pStart + os_strlen(needle);
}

/******************************************************************************
 * FunctionName : strncpy_null
 * Description  : copies up to n characters from the string pointed to by src to dest
 * Parameters   : dest
 *                src
 * Returns      : dest
 *******************************************************************************/
char ICACHE_FLASH_ATTR *strncpy_null(char *dest, char *src, uint16 n) {
	os_strncpy(dest, src, n);
	*(dest + n) = '\0';
	return dest;
}
/******************************************************************************
 * FunctionName : webserver_basic_authentication
 * Description  : check for HTTP basic authorization
 * Parameters   : pData -- HTTP request
 * Returns      : boolean
 *******************************************************************************/
LOCAL bool ICACHE_FLASH_ATTR webserver_basic_authentication(char *pData) {
	char *pStart = NULL;
	char *pEnd = NULL;

	int credentialsLen = 0;

	pStart = (char *) strstr_end(pData, "Authorization: Basic ");
	if (pStart == NULL) {
		return false;
	}

	pEnd = (char *) os_strstr(pStart, "\r\n");
	if (pEnd == NULL) {
		return false;
	}

	credentialsLen = pEnd - pStart + 1;

	char b64Credentials[credentialsLen];

	char credentials[credentialsLen];
	os_memset(credentials, '\0', sizeof(credentials));

	char user[credentialsLen];
	char passwd[credentialsLen];

	strncpy_null(b64Credentials, pStart, pEnd - pStart);

	int b64Res = base64_decode(b64Credentials, os_strlen(b64Credentials),
			credentials, &credentialsLen);

	pStart = (char *) os_strstr(credentials, ":");
	if (pStart == NULL) {
		return false;
	}

	strncpy_null(user, credentials, pStart - credentials);
	strncpy_null(passwd, pStart + 1,
			os_strlen(credentials) - (pStart - credentials));
	return webserver_authentication(user, passwd);
}` 

and in the webserver_recv I added this authentication check, I know the basic authentication is a joke which is why I am trying mbedtls to do better authentication.

`//os_printf(pusrdata);
    if(upgrade_lock == 0){

        // Check authentication
        if (sysCfg.httpd_auth){
        	if (!webserver_basic_authentication(pusrdata)) {
    			response_send(ptrespconn, 71);
    			return;
    		}
        }`

That aside, when I have this authentication enabled, the server crashes as mentioned in above post with the fatal exception. So, it looks like something going wrong with this authentication process.

Any idea why ?

FayeY, 2017-08-24:
Maybe due to the lack of memory? You can check it by system_get_free_heap_size.

crgor, 2017-08-24:
Finally, I found that the base64_decode call is causing the exception, the earlier SDK 1.5.2 was working fine with the method. 

After changing the method base64_decode to mbedtls_base64_decode, it worked fine. Sorry @FayeY for   misleading this issue as mbedtls issue, I should have checked compatibility of SDK2.1 before jumping into mbedtls.

closed 2017-08-24 by FayeY