espressif/ESP8266_NONOS_SDK #41: Missing IRAM_ATTR Macro?

alxAgu, 2017-09-17:
Hi, I was trying to use the supposed  IRAM_ATTR attribute mentioned in the "2c-esp8266_non_os_sdk" reference manual (p. 158). But after looking everywhere I couldn't find the definition. 

My c_types.h file doesn't seem to include it 

```
//NONOS_SDK, c_types.h
#ifdef ICACHE_FLASH
#define ICACHE_FLASH_ATTR __attribute__((section(".irom0.text")))
#define ICACHE_RODATA_ATTR __attribute__((section(".irom.text")))
#else
#define ICACHE_FLASH_ATTR
#define ICACHE_RODATA_ATTR
#endif /* ICACHE_FLASH */
```

while the RTOS_SDK does have it defined as:

```
//RTOS_SDK, c_types.h
#define IRAM_ATTR           __attribute__((section(".text")))
```

Am I missing something here?

Thanks.

A.

FayeY, 2017-09-18:
Hi, for ESP8266 non OS SDK, it needs not the IRAM_ATTR, the functions will be stored in the iRAM by default; and only with ICACHE_FLASH_ATTR the functions will be stored in the flash.

alxAgu, 2017-09-19:
Thanks @FayeY, I had no idea. I guess the documentation needs to be updated to mention this then?

A.

FayeY, 2018-09-21:
Document has been updated. See [ESP8266_Non_OS_SDK_API_Reference](https://www.espressif.com/sites/default/files/documentation/2C-ESP8266_Non_OS_SDK_API_Reference__EN.pdf) Section 2.2.

closed 2018-09-21 by FayeY

pylover, 2022-02-08:
I can confirm the documentation has not been updated yet.