espressif/ESP8266_NONOS_SDK #42: No way to stop TCP server started with espconn_accept when it has active clients.

rzajac, 2017-09-22:
When calling espconn_delete(&tcp_server) i get error -5 (ESPCONN_INPROGRESS) meaning there are clients I have to disconnect first but there is no way to do that. 

I tried code below but it does not work:

```
sint8 ICACHE_FLASH_ATTR srv_stop(void)
{
  sint8_t result;
  remot_info *remot = NULL;
  uint8 count = 0;

  if (espconn_get_connection_info(&tcp_server, &remot, 0) == ESPCONN_OK) {
    struct espconn conn;

    conn.proto.tcp = (esp_tcp *) os_zalloc(sizeof(esp_tcp));
    if (!conn.proto.tcp) {
      return ESPCONN_MEM;
    }

    for (count = 0; count < tcp_server.link_cnt; count++) {

      conn.type = ESPCONN_TCP;
      conn.state = remot[count].state;
      conn.proto.tcp->remote_port = remot[count].remote_port;
      conn.proto.tcp->remote_ip[0] = remot[count].remote_ip[0];
      conn.proto.tcp->remote_ip[1] = remot[count].remote_ip[1];
      conn.proto.tcp->remote_ip[2] = remot[count].remote_ip[2];
      conn.proto.tcp->remote_ip[3] = remot[count].remote_ip[3];

      result = espconn_disconnect(&conn);
      DEBUG("IP STOP: " IPSTR" %d:%d (%d)\n", IP2STR(conn.proto.tcp->remote_ip), conn.proto.tcp->remote_port, result);
    }

    os_free(conn.proto.tcp);
  }

  result = espconn_delete(&tcp_server);

  return result;
}
```

davydnorris, 2017-10-04:
try adding espconn_abort(&conn) as well to force the disconnect

rzajac, 2017-10-06:
@davydnorris I ended up tracking TCP connections. But this still does not really work as expected. The server stops listening (I'm not getting the callbacks) but it still accepts connections. You can see my solution with comments: https://github.com/rzajac/esp-cmd/blob/master/esp_cmd/esp_cmd.c#L280