espressif/ESP8266_NONOS_SDK #46: New release with the WPA2 fix

marcelstoer, 2017-10-17:
Please create a new release which includes the WPA2 fix b762ea222ee94b9ffc5e040f4bf78dd8ba4db596. I guess that would mean releasing from the tip of master to include all 13 commits as per https://github.com/espressif/ESP8266_NONOS_SDK/compare/v2.1.0...master.

jj2497, 2017-10-21:
It seems the fix results in reset wifi status. I can see the following messages periodically (I set ESP8266 to STATION mode only).
`
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt

connected with ssid-xxxxx, channel 8
dhcp client start...
`
If I revert back to SDK 2.1.0, it performs all right.

marcelstoer, 2017-10-30:
It's been nearly two weeks without any feedback. What's the status @espressif? I know that at least the NodeMCU project is waiting for a new release so we can ship a new release ourselves.

FayeY, 2017-10-31:
Hi, the latest ESP8266_NONOS_SDK fixed the WPA2 KRACK issue, and next ESP8266_NONOS_SDK release will be in the December. Thanks for your interest in ESP8266!

marcelstoer, 2017-10-31:
I guess the question is why you are not releasing a 2.1.1 immediately which includes the KRACK fix? Why did you do this for ESP-IDF but not for this SDK?

shinji2009, 2017-11-02:
sorry for spupid question i am not understand is crack fix already implemented on  https://nodemcu-build.com/ if i download master ?

marcelstoer, 2017-11-02:
@shinji2009 Unfortunately not. First Espressif has to release a new SDK (that's what this issue is about) then we need to incorporate it into the NodeMCU firmware (nodemcu/nodemcu-firmware#2138) and release it.
-> we're all waiting for Espressif

igrr, 2017-11-02:
Unfortunately we are a bit constrained by QA resources at this point, with multiple SDK releases which need testing (IDF 2.1.1, IDF 3.0, ESP8266 SDKs, ...). We will try make the release as soon as possible.

marcelstoer, 2017-11-02:
Thank you Ivan, that's sounds more promising than the previous response from Espressif 😜

wujiangang, 2017-11-03:
hi, @marcelstoer please try branch release/v2.1.x, which added KRACK fix. v2.2.0 will be arranged to test soon.

marcelstoer, 2017-12-30:
Any updates on this? The whole reason why I was asking for an official 2.1.x release was that we (NodeMCU) can pull it in automatically. Our build scripts look for `https://github.com/espressif/ESP8266_NONOS_SDK/archive/v$(SDK_FILE_VER).zip`. Hence, if we had wanted to pull in the fix from the branch we'd have had to temporarily adjust the build scripts. In hind sight maybe we should have... Is there an ETA for 2.2?

jcack-me, 2018-01-03:
We've also been patiently waiting on an official 2.2 release for several months. 
An ETA will help our release planning and help us to manage customer expectations.
thanks

jcack-me, 2018-01-03:
Many thanks, fingers crossed it doesn't slip!

marcelstoer, 2018-02-05:
> fingers crossed it doesn't slip!

Pity it didn't help 😢

jcack-me, 2018-02-05:
Hi FayeY,

The end of January has come and gone, and there is still no SDK release.
Do you have an update for us?

thanks

jcack-me, 2018-02-12:
Another week goes by and still we are left waiting ... :(

FayeY, 2018-02-12:
Hi, sorry that I thought there was a notification, the V2.2 has already released on github [Releases](https://github.com/espressif/ESP8266_NONOS_SDK/releases)

closed 2018-02-12 by FayeY

ESP-Geek, 2018-04-12:
@marcelstoer - Has this been integrated into the latest NodeMCU Build yet? 
Many thanks all for your hard work!

marcelstoer, 2018-04-12:
Yep, https://github.com/nodemcu/nodemcu-firmware/releases. Follow NodeMCU on FB at https://www.facebook.com/NodeMCU/ and never miss an important announcement.