espressif/ESP8266_NONOS_SDK #50: Issue with link ID

Boodorgad, 2017-10-25:
Problem:
If a Server is running and i receive a new Connection when a CIPSTART is also active(and tries to open a Connection to a host which doesn't exist), any new incomming TCP connection to the Server will use the same Link ID i used with CIPSTART.

simplified Setup:
CIPMUX=1
CIPSERVER=1,80
CIPSTART=**0**,"TCP","192.168.2.66",80

this may take some sec because 192.168.2.66 is not active.
Now connect to the Server an you get

**0**,CONNECT.....

At this Point the ESP won't work again.
There's no Returnvalue on the CIPSTART, and i just get the "busy p..." response on all further AT-commands. 

AT Version is 2.1.0

xcguang, 2017-11-29:
Hi @Boodorgad ,
Thanks for your report，we will reproduce and fix it.

MaJerle, 2017-11-30:
@Boodorgad If you have server enabled, I suggest you start your client connections from the biggest connection down, so 4 in this case. Server will populate 0, 1,... and you 4 therefore you can avoid collisions.

This is only temporary solution.

xcguang, 2017-12-01:
@Boodorgad @MaJerle Hi, yes, this is a bug. We will fix it next release.

FayeY, 2017-12-01:
Hi, we have already fixed it, but have not been released. We are working on the next release now, it is coming soon. Thanks for your interest in the ESP8266!

MaJerle, 2017-12-01:
We still have possible collision, no matter what we do. If somebody makes a connection active (as server) when string "AT+CIPSTART=0,...", we have collision.

The solution for this would be to simply send AT+CIPSTART="TYPE","IP",port and then ESP returns either result which connection was assigned to us or ERROR if no available connections

commit d6fa6dff (2017-12-05): fix(at): Fix that AT+CIPSTART causes busy if the server is connected at the same time

internal gitlab: c217922d

fix #50
closed 2017-12-05 by wujiangang