espressif/ESP8266_NONOS_SDK #53: IOTDemo Firmware OTA fails with mbedtls

crgor, 2017-11-02:
Hi All,

I have been testing Firmware OTA on IOTDemo with SSL enabled web server, I am using NON_OS SDK 2.1.0 with mbedtls library for ssl.

After doing the HTTPS post of https://<ipaddress>/config?command=start

I tried to upload the user bin file with curl command 

curl.exe -v -k -X POST --data-binary @D:\Reddy\Source_Installations\curl-7.46.0-win64\bin\$upgradefilename https://hostip
 
It fails with the message on the CURL output : 

Note: Unnecessary use of -X or --request, POST is already inferred.
* Rebuilt URL to: https://192.168.3.103/
*   Trying 192.168.3.103...
* Connected to 192.168.3.103 (192.168.3.103) port 443 (#0)
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / AES256-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*        subject: O=TLS Project; CN=127.0.0.1
*        start date: Aug 21 04:57:44 2017 GMT
*        expire date: Apr 30 04:57:44 2031 GMT
*        issuer: O=TLS Project Dodgy Certificate Authority
*        SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* Server auth using Basic with user 'admin'
> POST / HTTP/1.1
> Host: 192.168.3.103
> User-Agent: curl/7.46.0
> Accept: */*
> Content-Length: 371076
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
>
* Done waiting for 100-continue
* TLSv1.2 (IN), TLS alert, Client hello (1):
* SSL_write() returned SYSCALL, errno = 10053
* Closing connection 0
curl: (55) SSL_write() returned SYSCALL, errno = 10053

On the ESP command output side output as shown below:
dhcp client start...
ip:192.168.3.103,mask:255.255.255.0,gw:192.168.3.1
Free heap size = 35632
please start sntp first !
server handshake start.
server handshake ok!
Free heap size = 23744
len:224
A_dat:0,tot:0,lenght:0
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.3.33:52873 err 8 reconnect
pm open,type:2 0
server handshake start.
server handshake ok!
upgrade file download start.
Free heap size = 24744
server's data invalid protocol
Reason:[-0x7200]
webserver's 192.168.3.33:52876 err -114 reconnect
Free heap size = 35088
Free heap size = 35088
Free heap size = 35088

The output clearly shows the upgrade file download start.
Has anyone tested this?

kriegste, 2017-11-03:
```
server's data invalid protocol
Reason:[-0x7880]
```

is something I get, too. I have no idea what the SDK is trying to tell me...

marcelstoer, 2017-11-03:
> server's data invalid protocol

Ha, that looks familiar. Over at nodemcu/nodemcu-firmware#1707 a few people, namely @djphoenix and @nwf, have attempted to fix SSL issues that started with SDK 2.0.

kriegste, 2017-11-04:
The message I posted is displayed when esp works as an https server, not client.

The best I could achieve was one connection at a time. The message appears nonetheless. But as long as I make one connection after the other, it seems to work. But heaven forbid a disaster happens if I dare to attempt a connection while the old connection is still active! That breaks everything and I have to reboot. There is no exception, but following connections just "hang". It is more or less just like in the first post, I guess.

nwf, 2017-11-04:
"server's data invalid protocol" is, unfortunately, IIRC, emitted by the ESP firmware components as part of the ordinary exchange of shutdown messages, which is endlessly confusing.

zhang-jun-hao, 2017-11-13:
Hi @kriegste 

If client actively disconnected from ESP8266 (SSL/TLS server), it will report these two lines:
server's data invalid protocol
Reason:[-0x7880]
Reason[-0x7880] means the peer notified us that the connection is going to be closed.

Only one TCP connection is supported when using the current mbedtls library as TLS/SSl server. Please add MEMP_NUM_TCP_PCB = 1 before calling the espconn_secure_accept function.

Thanks

crgor, 2017-11-13:
Zhang,

Thanks for the reply.

You mean to say, when I am upload the binary file with curl the connection
is disconnected ?
What is the default value of MEMP_NUM_TCP_PCB when running as SSL server ?

Would keep alive helps on the curl uploading ?

Regards,Reddy.

zhang-jun-hao, 2017-11-13:
Hi  @crgor 

The default value of MEMP_NUM_TCP_PCB is 5 witch is set in function lwip_init()
  MEMP_NUM_TCP_PCB = 5;
#define MEMP_NUM_TCP_PCB                (*(volatile uint32*)0x600011FC)

You made a mistake in making the certificate. 
Please take a closer look at the document in this link:
http://www.espressif.com/en/support/download/documents?keys=ESP8266+Non-OS+SDK+SSL+User+Manual

Change the IP address 127.0.0.1 to 192.168.3.103 before run makefile.sh.

Thanks!

kriegste, 2017-11-13:
IIRC this should do the same:
espconn_tcp_set_max_con(1);

This will affect all other TCP servers, too. :(

But it is better (no exception anymore).
However, after hammering F5 for a while I noticed that the sent callback is no longer called. This might be a totally new problem. I will try to investigate and report.

crgor, 2017-11-13:
@zhang-jun-hao 

Yes, that is indeed what I did, I will try tomorrow morning with CN as IP address of SSL server.

But what I don't understand is that I have disabled SSL verification on the Client side. I thought that should not be a problem. 

Anyhow, I have many ESP's to Flash, I cant use Static IP's for generating certificates. May I use Hostname there?

crgor, 2017-11-14:
@zhang-jun-hao 
I have defined MEMP_NUM_TCP_PCB as 1 and regenerated certs with CN as 192.168.3.104 which is the static IP.

Serial output : 
=== System init ====
mode : sta(60:01:94:84:0d:2e)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

connected with IOTLab, channel 8
dhcp client start...
ip:192.168.3.104,mask:255.255.255.0,gw:192.168.3.1
Free heap size = 35632
please start sntp first !
Free heap size = 35632
pm open,type:2 0
Free heap size = 35120
server handshake start.
server handshake ok!
len:193
A_dat:0,tot:0,lenght:0
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.3.33:35531 err 8 reconnect
Free heap size = 35088
server handshake start.
server handshake ok!
upgrade file download start.
Free heap size = 22648
server's data invalid protocol
Reason:[-0x7200]
webserver's 192.168.3.33:35532 err -114 reconnect
Free heap size = 35088

Curl output

Note: Unnecessary use of -X or --request, POST is already inferred.
*   Trying 192.168.3.104...
* Connected to 192.168.3.104 (192.168.3.104) port 443 (#0)
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / AES256-SHA256
* ALPN, server did not agree to a protocol
* Server certificate:
*        subject: O=TLS Project; CN=192.168.3.104
*        start date: Nov 14 04:18:18 2017 GMT
*        expire date: Jul 24 04:18:18 2031 GMT
*        issuer: O=TLS Project Dodgy Certificate Authority
*        SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* Server auth using Basic with user 'admin'
> POST / HTTP/1.1
> Host: 192.168.3.104
> Authorization: Basic YWRtaW46YWRtaW4=
> User-Agent: curl/7.46.0
> Accept: */*
> Content-Length: 371220
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
>
* Done waiting for 100-continue
* TLSv1.2 (IN), TLS alert, Client hello (1):
* SSL_write() returned SYSCALL, errno = 10053
* Closing connection 0
curl: (55) SSL_write() returned SYSCALL, errno = 10053

I tried with Keep alive, didn't made any difference. I suspect it is something to do with the connection being broken after the first packet of the binary file is uploaded. curl is closing the connection for some reason.
I am going to try one more thing, CA enabled and make sure this whole business of the SSL handshake and encryption works smoothly first.

@zhang-jun-hao  any tips?

crgor, 2017-11-14:
Tried with CA_Enabled on ESP SSL server 

Serial output:
mode : sta(60:01:94:84:0d:2e)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

connected with IOTLab, channel 8
dhcp client start...
Free heap size = 35600
WiFi connecting...
ip:192.168.3.104,mask:255.255.255.0,gw:192.168.3.1
Free heap size = 35600
please start sntp first !
pm open,type:2 0
Free heap size = 34576
please start sntp first !
server handshake start.
espconn_mbedtls.c 652, type[TLS.ca_x509.cer],length[479]
server handshake ok!
len:138
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.3.101:58652 err 8 reconnect
Free heap size = 35056
please start sntp first !
Free heap size = 35056
please start sntp first !
server handshake start.
espconn_mbedtls.c 652, type[TLS.ca_x509.cer],length[479]
server handshake ok!
upgrade file download start.
server's data invalid protocol
Reason:[-0x7200]
webserver's 192.168.3.101:58654 err -114 reconnect
Free heap size = 35056
please start sntp first !

Curl Output:

curl -v --cacert TLS.ca_x509.pem --cert TLS.x509_1024.pem --key TLS.key_1024.pem --key-type PEM -u admin:admin -X POST https://192.168.3.104/upgrade?command=start
*   Trying 192.168.3.104...
* Connected to 192.168.3.104 (192.168.3.104) port 443 (#0)
* found 1 certificates in TLS.ca_x509.pem
* found 758 certificates in /etc/ssl/certs
* ALPN, offering http/1.1
* SSL connection using TLS1.2 / RSA_AES_256_CBC_SHA256
* 	 server certificate verification OK
* 	 server certificate status verification SKIPPED
* 	 common name: 192.168.3.104 (matched)
* 	 server certificate expiration date OK
* 	 server certificate activation date OK
* 	 certificate public key: RSA
* 	 certificate version: #1
* 	 subject: O=TLS Project,CN=192.168.3.104
* 	 start date: Tue, 14 Nov 2017 04:18:18 GMT
* 	 expire date: Thu, 24 Jul 2031 04:18:18 GMT
* 	 issuer: O=TLS Project Dodgy Certificate Authority
* 	 compression: NULL
* ALPN, server did not agree to a protocol
* Server auth using Basic with user 'admin'
> POST /upgrade?command=start HTTP/1.1
> Host: 192.168.3.104
> Authorization: Basic YWRtaW46YWRtaW4=
> User-Agent: curl/7.43.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Content-Length: 0
< Server: lwIP/1.4.0
< 
* Closing connection 0
rerddy@rerddy-VirtualBox:~/code/ESPFirmware$ curl -v --cacert TLS.ca_x509.pem --cert TLS.x509_1024.pem --key TLS.key_1024.pem --key-type PEM -u admin:admin -X POST --data-binary @user1_SonoffT13CH.bin  https://192.168.3.104/
*   Trying 192.168.3.104...
* Connected to 192.168.3.104 (192.168.3.104) port 443 (#0)
* found 1 certificates in TLS.ca_x509.pem
* found 758 certificates in /etc/ssl/certs
* ALPN, offering http/1.1
* SSL connection using TLS1.2 / RSA_AES_256_CBC_SHA256
* 	 server certificate verification OK
* 	 server certificate status verification SKIPPED
* 	 common name: 192.168.3.104 (matched)
* 	 server certificate expiration date OK
* 	 server certificate activation date OK
* 	 certificate public key: RSA
* 	 certificate version: #1
* 	 subject: O=TLS Project,CN=192.168.3.104
* 	 start date: Tue, 14 Nov 2017 04:18:18 GMT
* 	 expire date: Thu, 24 Jul 2031 04:18:18 GMT
* 	 issuer: O=TLS Project Dodgy Certificate Authority
* 	 compression: NULL
* ALPN, server did not agree to a protocol
* Server auth using Basic with user 'admin'
> POST / HTTP/1.1
> Host: 192.168.3.104
> Authorization: Basic YWRtaW46YWRtaW4=
> User-Agent: curl/7.43.0
> Accept: */*
> Content-Length: 371220
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
> 
* Done waiting for 100-continue
* Closing connection 0
curl: (55) Failed sending data to the peer

I am out of options to test

crgor, 2017-11-16:
@FayeY @wujiangang 
I have tested IOT_Demo OTA with mbedtls and ssl libraries both have problem, just checking have you guys tested on your side? I am eager to find out what am I doing wrong

zhang-jun-hao, 2017-11-16:
Hi @crgor,

Mbedtls can only receive 5 x 1460 bytes at a time. The decryption needs to receive the entire bin file. So it generates error when you send more than 20K bin file. The bin file cannot exceed 4K when upgrading with the mbedtls library.

Here is the test log I used with a 4K bin file:

curl -kv -d -X POST  https://192.168.4.1/upgrade?command=start
curl -kv -X POST --data-binary @./4K.bin  https://192.168.4.1

server handshake start.
server handshake ok!
A_dat:2,tot:2,lenght:2
We have a POST request.
local upgrade start
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.4.2:38126 err 8 reconnect
server handshake start.
server handshake ok!
upgrade file download start.
upgrade file download finished.
local upgrade success
system upgrade deinit
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.4.2:38128 err 8 reconnect

zhang-jun-hao, 2017-11-16:
Hi @crgor 

Embedtls can only receive 5*1460 bytes data at a time. 
Data decryption needs to receive a complete bin file. 
So sending more than 20K bytes of bin files goes wrong.

Here is the result of my test with a 4K bin file:

PC:
curl -kv -d -X POST  https://192.168.4.1/upgrade?command=start
curl -kv -X POST --data-binary @./4K.bin https://192.168.4.1

ESP8266:
server handshake start.
server handshake ok!
A_dat:2,tot:2,lenght:2
We have a POST request.
local upgrade start
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.4.2:51420 err 8 reconnect
server handshake start.
server handshake ok!
upgrade file download start.
upgrade file download finished.
local upgrade success
system upgrade deinit
server's data invalid protocol
Reason:[-0x7880]
webserver's 192.168.4.2:51422 err 8 reconnect

crgor, 2017-11-26:
I was able to make it work, I had to tweak a little bit on the ESP local_upgrade_download function and write a java HTTP file uploader(This uploads each 8Kbytes chunks as a new HTTP post request).

It takes nearly a minute to upgrade a firmware file size of 370Kbytes.

@wujiangang any way I could improve the speed by increasing the packet size on mbedtls library ?

Alexis-Polegato, 2017-12-06:
Hello @zhang-jun-hao 

Regarding this issue; do you know if there will be a fix or a workaround to be able to dowload a bin file from a Secure Server with the ESP8266 or we need to dowload it manually in several packets as suggested @crgor  ?

Thanks for your update.