espressif/ESP8266_NONOS_SDK #54: Bug in API reference

marcelstoer, 2017-11-08:
Chapter 3.3.13 at http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf#page=31 has the wrong title.

Wrong: system_phy_set_rfoption (that's for chapter 3.3.11)
Correct: system_phy_set_max_tpw

kriegste, 2017-11-09:
I have one, too.

http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf#page=126
4.2.15. espconn_regist_write_finish
The description is bogus (copy/paste).

wujiangang, 2017-11-28:
@marcelstoer @kriegste Thank you! 

@FayeY Please modify the document.

FayeY, 2017-11-28:
Hi, thanks for your reporting! The document is updated, see [ESP8266 Non-OS SDK API Reference](http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf). Thanks.

closed 2017-11-28 by FayeY