espressif/ESP8266_NONOS_SDK #57: PWM starts working on 2.1.0 but breaks WIfi AP hosting

AgentSmithers, 2017-11-26:
Hi Everyone.
I am using the Wemos D1 Mini Pro and I came across the requirement to get PWM working. So I went into my main SDK directory and modified the Makefile adjusting it from VENDOR_SDK = 2.0.0 to VENDOR_SDK = 2.1.0 and BAM PWM started working perfect without my WD timer kicking off.. But then right when I did that my Wifi stoped working. It seems like everything is perfect but it is no longer hsoting an AP anymore, I then updated my make file back to 2.0.0 then did a clean and make on my project then Wifi starts working then my PWM kicks off a WDT error rebooting my chip. Any qlues? I can reproduce the issue swapping back and forth to make the Wifi or PWM stop working correctly.
Thank you everyone!

# Whether to merge SDK into Xtensa toolchain, producing standalone
# ESP8266 toolchain. Use 'n' if you want generic Xtensa toolchain
# which can be used with multiple SDK versions.
STANDALONE = y

# Directory to install toolchain to, by default inside current dir.
TOOLCHAIN = $(TOP)/xtensa-lx106-elf

# Vendor SDK version to install, see VENDOR_SDK_ZIP_* vars below
# for supported versions.
VENDOR_SDK = 2.0.0

.PHONY: crosstool-NG toolchain libhal libcirom sdk

TOP = $(PWD)
SHELL = /bin/bash
PATCH = patch -b -N
UNZIP = unzip -q -o
VENDOR_SDK_ZIP = $(VENDOR_SDK_ZIP_$(VENDOR_SDK))
VENDOR_SDK_DIR = $(VENDOR_SDK_DIR_$(VENDOR_SDK))

VENDOR_SDK_DIR_2.1.0-18-g61248df = ESP8266_NONOS_SDK-2.1.0-18-g61248df
VENDOR_SDK_ZIP_2.1.0 = ESP8266_NONOS_SDK-2.1.0.zip
VENDOR_SDK_DIR_2.1.0 = ESP8266_NONOS_SDK-2.1.0
VENDOR_SDK_ZIP_2.0.0 = ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip
VENDOR_SDK_DIR_2.0.0 = ESP8266_NONOS_SDK_V2.0.0_16_08_10
VENDOR_SDK_ZIP_1.5.4 = ESP8266_NONOS_SDK_V1.5.4_16_05_20.zip
VENDOR_SDK_DIR_1.5.4 = ESP8266_NONOS_SDK_V1.5.4_16_05_20
VENDOR_SDK_ZIP_1.5.3 = ESP8266_NONOS_SDK_V1.5.3_16_04_18.zip
VENDOR_SDK_DIR_1.5.3 = ESP8266_NONOS_SDK_V1.5.3_16_04_18/ESP8266_NONOS_SDK
VENDOR_SDK_ZIP_1.5.2 = ESP8266_NONOS_SDK_V1.5.2_16_01_29.zip
VENDOR_SDK_DIR_1.5.2 = esp_iot_sdk_v1.5.2

FayeY, 2017-11-28:
Hi @AgentSmithers , could you provide a project with your test code, so that we can reproduce the problem and help debug it?

AgentSmithers, 2017-11-29:
Yes sir FayeY, Both are attached, The PWMI2C is a very simple project. it just starts PWM and the I2C is basically disabled, Then the other project is a bit complicated but really it just host a AP on the chip at the start and doesn't broadcast at all unless I roll back to SDK 2.0.0
The chip I am using is here
https://wiki.wemos.cc/products:d1:d1_mini_pro

Thank you again! Very appreciated.
[Sources.tar.gz](https://github.com/espressif/ESP8266_NONOS_SDK/files/1512560/Sources.tar.gz)

AgentSmithers, 2017-12-04:
Hi FayeY, Any luck, anthing I can do to assist. Im sadly stuck in the water.

ESP-Coco, 2017-12-05:
Hi @AgentSmithers ,  sorry reply so late. I tested your PWMI2C project and add AP mode, i found you use    the api os_delay_us to delay sometime. This will suspend the system so many time. I also found it is used in the second project.   I think this may be the reason why it cannot work. I changed it using os_timer to delay and tested it, I found both SDK 2.0 and 2.1 work well . I upload my modified file ,Hope to help you :)
[PWMI2C.tar.gz](https://github.com/espressif/ESP8266_NONOS_SDK/files/1530279/PWMI2C.tar.gz)

AgentSmithers, 2017-12-05:
Hi @ESP-Coco, No problem sir. Where you able to reproduce the broken state? Heck I will try your recommendation and if it work's I owe you Coffee sir! :) Any idea why os_delay_us breaks the PWM? Is there some internal mechanism that must be kicked off at a specific time where os_delay impacts to kick off the pwm issue? Thank you so much for your help!

kriegste, 2017-12-05:
For how long did you try to halt the system? You cannot halt it for too long since NONOS is no real time OS. It needs opportunities to handle its background tasks and the wifi functions are time-critical.

AgentSmithers, 2017-12-05:
Hi @kriegste , Thank you for chiming in. Would changing from 2.0 to 2.1 effect it in that way, it seems OS_Delay was very very solid in 2.0 but when moving to 2.1 it stopped accepting arguments over a 16bit value, I think and was not delaying at all unlike 2.0 where I can call os_delay_us(5 * 1000 * 1000) for 5 seconds (No issues at all with this). After when moving to 2.1 I find that it does not sleep at all if the value is over 65535 (Don't qutote me on this value but I think that was the unsigned16bit) so I had to create a custom function to delay a full second.

`void delay_second()
{
	system_soft_wdt_feed();
	long i = 0;
	for (i;i<1000;i++)
	{	
		os_delay_us(1000);
	}
	system_soft_wdt_feed();
	return;
}`

Again, this works perfectly fine in 2.0 but when moving to 2.1 it appears I have to drop the function above, then Wifi stops operating.

Thank you all for your speedy feedback.

marcelstoer, 2017-12-05:
The answer is here: http://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf#page=41

> max : 65535 μs

-> Espressif doesn't want you to stop the ESP8266 world for long; if you do the WiFi stack will fail

kriegste, 2017-12-05:
Correct. And exceeding limits causes undefined behavior, so differently acting SDK versions would be no surprise.

AgentSmithers, 2017-12-05:
Gotcha, Ill push 2.1 to my currently Wifi project and swap out os_delays to use Timers.

AgentSmithers, 2017-12-12:
Well this is the weirdest thing, I upgraded back to 2.1 and did a full ROM erase and reupload which I've done before and now it's working. Very weird. I will test further. Thank you, everyone.

AgentSmithers, 2017-12-14:
Team, Once the firmware is burned, Everything looks great. After reboot the Wifi then no longer functions.
After adding

wifi_set_sleep_type( NONE_SLEEP_T );

To the start of the project things are now operating correctly with 2.1.0

I assume something changed in the firmware that requires this to be called unlike 2.0.0

I will continue testing but it looks clear so far.

closed 2017-12-25 by FayeY