espressif/ESP8266_NONOS_SDK #58: ifi_station_set_enterprise_username  does not seem to set user name.

penright, 2017-11-27:
I could not get WPA2 to connect,  [then I found this article](https://www.hallgeirholien.no/post/esp8266-eap/). In it the author describes how to hack the libwpa2.a. Once I did hack the libwpa2.a file, I was able to connect to WPA2 Enterprise. If I can try anything to help debug please let me know.

Here is my code
```

#include <Arduino.h>
#include <ESP8266WiFi.h>
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}
#endif

static u8 ent_username[] = "MyUserName";
static u8 ent_password[] = "MyPassword";

void setup(){
  char a[100];
  ip_info info;
  wifi_get_ip_info(0, &info);

  Serial.begin(115200);

  wifi_station_disconnect();
  wifi_set_opmode(STATION_MODE);

  char ssid[32] = "MySSID";
  char password[64] = {0x00};
  struct station_config stationConf;
  stationConf.bssid_set = 0;  //need not check MAC address of AP
  memcpy(&stationConf.ssid, ssid, 32);
  memcpy(&stationConf.password, password, 64);

  if(!wifi_station_set_config(&stationConf)){
    Serial.print("\r\nset config fail\r\n");
  }

  // switch to WPA2 Enterprise 
  wifi_station_set_wpa2_enterprise_auth(1); 

  if(wifi_station_set_enterprise_username (ent_username, sizeof(ent_username))){
    Serial.print("\r\nusername set fail\r\n");
  }
  if(wifi_station_set_enterprise_password (ent_password, sizeof(ent_password))){
    Serial.print("\r\npassword set fail\r\n");
  }

  if(!wifi_station_connect()){
    Serial.print("\r\nconnect fail\r\n");
  }

  Serial.print("\r\ntrying to connect...");

  while(info.ip.addr == 0){
    ESP.wdtFeed();
    Serial.print(".");
    delay(1000);
    wifi_get_ip_info(0, &info);
  }

  sprintf(a, "%"PRIu32,info.ip.addr);
  Serial.print("\r\nip addr: ");
  Serial.print(a);
  Serial.print("\r\n");
}

void loop()
{ 
}
```

kriegste, 2017-11-27:
Maybe use
int wifi_station_set_enterprise_identity(u8 *identity, int len);
which was added some weeks ago?

wujiangang, 2017-11-28:
thanks for your answer, @kriegste 

@penright I'm not sure whether arduino has used master branch of nonos sdk, you can copy the libwpa2.a in the master branch to arduino platform, and have a try.

penright, 2017-11-28:
Thanks @kriegste and @wujiangang 

I trying to find the answer to what branch nonos sdk Arduino uses. Right now I am deconstruction the Arduino. I think it does use the master branch.
Since it does not look like a bug and while I am digging should we close this? 
Is there a forum I could post to? The ones I have tried so far do not seem very actived.

While I am here, can I ask a point of clarification....
So the hello world example I was using had a "username" (wifi_station_set_enterprise_username) and "password" (wifi_station_set_enterprise_password) function. Which is what my phone asked logging into the WPA2 Enterprise (MySSID). So is  wifi_station_set_enterprise_identity used instead of wifi_station_set_enterprise_username?

penright, 2017-11-29:
I found the answer to one question. Here is  [the change log](https://github.com/esp8266/Arduino/blob/master/doc/changes.rst#changelog) for the Arduino board manager that installs the esp8266.
If I am reading it right, Version 2.3.0 (Arduino ?), June 2016, SDK 1.5.3 (esp8266 ?) .

penright, 2017-11-29:
One last shot I tried to copy the libwpa2.a and the wpa2_enterprise.h files into my libs.
Then I add the set identity and all compiled. But when linking I get a **_`.text' will not fit in region `iram1_0_seg'_** By commenting out different lines I have narrowed it to setting the enterprise auth to 1 "wifi_station_set_wpa2_enterprise_auth(1)" I did not get this link error before the hack.

Sounds like I need to wait till @igrr updates the board manager.
If wrong maybe he will correct me.

victorclaessen, 2017-12-25:
Hi penright, I had the same problem as you, see [this comment](https://github.com/esp8266/Arduino/issues/1032#issuecomment-314477339). The way I got it to work is [here](https://github.com/esp8266/Arduino/issues/1032#issuecomment-321812646).

Now if they could only fix the [EAP-selection bug](http://bbs.espressif.com/viewtopic.php?p=18595&sid=e8e14cd6f97ee908d96ca75d0d47edd8#p15165), we would all be on our way.

penright, 2017-12-25:
@victorclaessen Thanks. I found a blog that got me connected. 
[here](https://www.hallgeirholien.no/post/esp8266-eap/)
Looks like the same steps.
The issue now is now way to edit the user name. The only way I could connect was to binary edit the the object file. 
Here was his steps ...

> 3. Modify the file libwpa2.a (in …/packages/esp8266/hardware/esp8266/tools/sdk/lib):
> Using a hex editor(i.e. vi), replace the text anonymous@espressif.com with your mail address, fill the spare positions with . (periods) to ensure that the text length is not unchanged

I was trying to add it to the Open Garage WiFi gui. Without the username being soft I am still stuck.

victorclaessen, 2017-12-25:
So in the time since the blog post that you mention, ExpressIf have added the _wifi_station_set_enterprise_identity_ function to the API. So now you do not need to edit libwpa2.a anymore. You can just set the identity with that function.
But you _will_ need the new version of the ExpressIf SDK (update_sdk_2.1.0). For details, see my posts.