espressif/ESP8266_NONOS_SDK #59: AT 2.1 AT data not in sequence

MaJerle, 2017-11-29:
Hello team,

I've working daily now with ESP8266 and I'm using AT software 2.1. Today I found very strange behavior of AT software when returning data from device to my MCU.

```
0,CONNECT

+IPD,0,326,192.168.1.3,5285:GET /js1.js HTTP/1.1
Host: 192.168.1.26
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/62.0.3202.94 Safari/537.36
Accept: */*
Referer: http://192.168.1.26/
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8,sl;q=0.7

AT+CIPSEND=0,1,CONNECT
98

OK
> 
+IPD,1,345,192.168.1.3,5286:GET /style1.css HTTP/1.1
Host: 192.168.1.26
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/62.0.3202.94 Safari/537.36
Accept: text/css,*/*;q=0.1
Referer: http://192.168.1.26/
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en;q=0.9,en-US;q=0.8,sl;q=0.7

Recv 98 bytes

SEND OK
AT+CIPCLOSE=0
0,CLOSED

OK
1,CLOSED
```

Please focus on a line:
```
AT+CIPSEND=0,1,CONNECT
98
```
Looks like I received `1,CONNECT` in the middle of ECHO response `AT+CIPSEND=0,98`. 

Why is that? In this case I cannot determine actual CONNECT status if I receive random data!

Thank you for your support.

FayeY, 2017-12-04:
Hi, yes, it is really strange, this problem should not happen. Could you please check your HOST MCU or check the data on UART, to see if there is any pause or gap between your "AT+CIPSEND=0," and the data length "98"? In fact, if you input the command "AT+CIPSEND=0,98" at once, the "1,CONNECT" should not be in the middle of it.

MaJerle, 2017-12-04:
Honestly, I send my data this way:

`AT+CIPSEND=`
`0`
`,`
`98`
`\r\n`

So all together 5 gaps are possible. Why is that? Why ECHO mode immediately returns character and does not wait for `<CR><LF>` first?

FayeY, 2017-12-05:
Hi yes, it is immediately returns, will not wait for the `<CR><LF>`. In case, there may not be a `<CR><LF>`. For example, some beginners may miss the `<CR><LF>` when they first use the AT commands, and if there is no echo it will make them feel more confuse. And if we set a timeout, then some users may think it echo too slowly. So we designed it to return immediately.

Arno-Nymous, 2017-12-05:
This doesn't make much sense to me. That's what a command documentation is for: making clear, how each command is exactly to be structured. If I ending some commands with CRLF is optional then this is much more confusing (all the more for beginners!) than it is of any help. Plus, such behavior nips making good software on top of the AT firmware in the bud. So I beg you: pleas3 correct this and make a CRLF mandatory for EACH command and make the ESP wait for it forever, if necessary. That way it will become ingrained even for beginners on the first day.

MaJerle, 2017-12-06:
@Arno-Nymous I fully agree with you. AT Commands are not only "for fun". I'm working on a pro library using AT commands and here is the reason I'm posting all the requests and observations I found under development.

FayeY, 2017-12-06:
Hi both, 
I'm not sure if there is any misunderstanding. Since English is not our first language, I may have difficulty with understanding long sentence. So please forgive that sometimes I may need you to explain more. Sorry for the inconvenience.
Here, I will explain my point first:
1. For execution, AT commands are always waiting for the `<CR><LF>`, it will wait it forever. For example, you input an "AT+GMR" first, and wait 10 minutes and then input the `<CR><LF>`, it will still response the version information when it gets the `<CR><LF>`.
2. For echo (display), which can be disabled/enabled by commands ATE0/ATE1, AT commands returns immediately, to let the users know what they have inputted if it is needed. 
I think what we discussing here is to make the echo also wait `<CR><LF>` forever. But sorry that I do not get why this is needed. I think the most common cases is that the Host MCU will disable the echo by ATE0.

MaJerle, 2017-12-06:
@FayeY Thank you for detailed info. If this is the case, I understand and we can close the loop.

closed 2017-12-06 by MaJerle