espressif/ESP8266_NONOS_SDK #60: SEND FAIL followed by SEND OK

MaJerle, 2017-11-29:
Hello team,

in the past few days I can constantly see that when I use `AT+CIPSEND` ESP returns with `SEND FAIL` followed by `SEND OK`. What is happening behind? I cannot know what is the correct response and if this happens, what is the maximal time between `SEND FAIL` and `SEND OK` responses?

Thank you for your support.

FayeY, 2017-12-04:
Hi, sorry that we cannot reproduce your problem, could you provide more details of your test steps? So that we can test it exactly as your way. And also please provide the full version information which got by the command AT+GMR. Thanks.

MaJerle, 2017-12-04:
`AT+GMR` shows latest version:

```
AT version:1.5.0.0(Oct 24 2017 12:03:18)<\r><\n>
SDK version:2.1.0(ace2d95)<\r><\n>
compile time:Oct 24 2017 15:48:02<\r><\n>
Bin version(Wroom 02):1.5.1<\r><\n>
```

I found later that this problem was reported by someone else too. Root cause of it would be that you somehow do resend if packet fails. If you do a resend, you should report `SEND FAIL` only if non of resends work.

FayeY, 2017-12-20:
Hi,
Sorry that we still cannot reproduce your problem.
Could you provide the entire test log for debugging? 
Thanks.

Maintraxx, 2018-11-02:
Bug is still present in SDK 3.0.0

FayeY, 2018-11-04:
@Maintraxx Could you provide the log from both UART1 and UART0 for debugging? Thanks in advance.

Maintraxx, 2018-11-22:
Here is it:
(Socket 0 is connected to a web browser, Socket 4 is used for the communication with openweathermap.org"

```AT+CIPSTART=4,"TCP","api.openweathermap.org",80

+LINK_CONN:0,4,"TCP",0,"95.85.63.65",80,9433

OK

AT+CIPSEND=4,173

OK
> 
GET /data/2.5/weather?appid=890A8DDF9AEEDD433E40C2CA3AAAAAAA&zip=12345,de&lang=de HTTP/1.1
Host: api.openweathermap.org
User-Agent: My ESP/1.0
Connection: close

Recv 173 bytes

SEND OK

AT+CIPSEND=0,988

OK
> 
#######988 BYTES OF BINARY DATA #########

Recv 988 bytes

+IPD,4,782
4,CLOSED
SEND FAIL

SEND OK

AT+CIPSEND=0,1656

OK
> ````

Maintraxx, 2019-05-10:
@FayeY @xcguang 
Here are more detailed information. The problem is also in the latest SDK 3.1.0 dev valid.

[image]

Here the SEND FAIl / SEND OK in detailed:

[image]

Here is the "full" context:

[image]

closed 2020-12-17 by MaJerle

Maintraxx, 2020-12-17:
Why did you close it? The problem is still exist!