espressif/ESP8266_NONOS_SDK #61: AT feature request: Set number of connections

MaJerle, 2017-11-29:
Feature request for AT. We all know that number of connections is only limited to RAM usage. It would be good if we have AT command to set maximal number of connections such as `AT+SETCONNNUM=x` where x means maximal number of connections.

And then `AT+CIPCLOSE=x` would close all the connections, with the same behavior as it works now.

FayeY, 2017-12-04:
Hi, sorry that I'm a little confused, is this a duplicate issue with #62 ? Or they are two different requests?

MaJerle, 2017-12-04:
Hello, as it is stated in description, this are 2 different proposals. One is setting maximal connections using AT command (max connections all together) while another one is setting max number of connections for server.

FayeY, 2017-12-04:
Hi, the max connections allowed can be set by adding a "at_customLinkMax=n" in user_init() before calling at_init().

MaJerle, 2017-12-04:
I know but it is not via official AT.

FayeY, 2017-12-05:
Hi, it is a feature in the official AT, you can add "at_customLinkMax=n" in the official AT. Default is 5.
It is not a command, because that the memory those links will take are pre-set in the source code, then be compiled.

MaJerle, 2017-12-05:
Cant we change this to dynamic memory usage?

FayeY, 2017-12-05:
Hi, sorry that it is not dynamic. To be dynamic, it needs a big change in our internal framework, but since the ESP8266 AT is a stable product, and many customers are using it. We prefer not to change it a lot. Thanks.

MaJerle, 2017-12-05:
Thanks. There is a catch. It is not very stable, I will post another AT feature request  with detailed explanation what happens.

closed 2017-12-06 by FayeY