espressif/ESP8266_NONOS_SDK #62: AT+CIPSERVERMAXCONN like in esp32-at

MaJerle, 2017-11-30:
Hello team,

In ESP32-AT there is a command `AT+CIPSERVERMAXCONN`. Is there a plan to integrate the same command here? It would be very useful to limit resources for SERVER only.

What about this implementation here? 
https://github.com/espressif/esp32-at/issues/28

FayeY, 2017-12-05:
Hi, we will add it in the next release. Thanks for all your valuable advices! :)

MaJerle, 2017-12-20:
If I have 5 connections all together and 5 connections are assigned to server. Is there a way to tell ESP to not allow client mode, so if I do `AT+CIPSTART=4,...` it returns error?

MaJerle, 2017-12-21:
Even if I have set MAXCONN to 10, server will still populate only up to 5 connections (0-4), even if I have 20 connections possible in ESP (compiled with my own settings). Something is hardcoded to 5 for some reason.

FayeY, 2017-12-22:
Hi

> server will still populate only up to 5 connections (0-4), even if I have 20 connections possible in ESP (compiled with my own settings). 

Have you tested it? This should not happen. If you add at_customerLinkMax=9; before at_init, then the ESP server can accept 9 links at most. 
Thanks.

MaJerle, 2017-12-22:
I tested it. If I open 7 connections to esp acting as server, only 0-4 are open, and I have fo close one to allow next connection ti be processed.

FayeY, 2017-12-22:
Hi, sorry that we cannot reproduce your problem.. Could you have a try with the user.bin I provided in [issue 77](https://github.com/espressif/ESP8266_NONOS_SDK/issues/77#issuecomment-353017624)

FayeY, 2017-12-22:
The test project we used is as the attachment, please try to compile and test it. If your problem is still unsolved, please provide your test project, we will have a try with it. Thanks.
[ESP8266_NONOS_SDK-master.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1581531/ESP8266_NONOS_SDK-master.zip)

MaJerle, 2017-12-22:
On the image below you can see how I opened 6 sockets from PC but only 5 of them were successful.

[image]

Project I used is also here (FLASH conf 2 (512+512)):
[user1.1024.new.2.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1581600/user1.1024.new.2.zip)

FayeY, 2017-12-22:
Have you set at_customerLinkMax=9; before at_init and then re-compile it? 
We have provided our test project in the comment above, you can compile it and have a try.
Thanks.

MaJerle, 2017-12-22:
When I try compile your example it compiles infinite time saying "user_main.d" is compiled with time in future. If I delete output files and start over, still a problem.

FayeY, 2017-12-22:
Hi, 
Sorry for the inconvenience.
After checking it with our engineer, it is a limitation in the LWIP [#define linkMax 15](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/include/lwip/app/espconn.h#L208).
You can:
1. change the `#define linkMax 15`  
2. then run `./make_lib.sh lwip` in the "ESP8266_NONOS_SDK/third_party" to re-build the LWIP  
3. then re-build the AT bin.  
Please note that too many connections may cause an issue of "out of memory". You can check the memory by command AT+SYSRAM?
Thanks.

closed 2018-04-16 by FayeY