espressif/ESP8266_NONOS_SDK #63: Extending x,CONNECT functionality

MaJerle, 2017-12-03:
Hello team,

it would be good, if we know immediately more info when a new client connects to device, such as IP address and port so we can react fast, not to wait AT+CIPSTATUS command.

Response something like:

`+CIPSTAT:num,c_s,ip,port` where:

`num`: Connection number where something happen
`c_s`: Number either connection is client or server
`ip`: Remote IP address
`port`: Remote port

FayeY, 2017-12-04:
Hi, thanks for your valuable advice!
We are going to add a command AT+CIPSTATUS=1/0 to enable/disable it.
If it is enabled, the message will be "<link_id>,<TCP/UDP>,<C/S>,<remote_ip>,<remote_port>,CONNECT"
And it is disable by default, the message is the original one, "<link_id>, CONNECT"
Thanks.

MaJerle, 2017-12-04:
@FayeY I would start with known value such as +SOMETHING as it makes it easier to parse because you have a known value you can lean on start.

Please discuss about this.

FayeY, 2017-12-04:
Hi @MaJerle , but if it is 
`
<link_id>, CONNECT
+CIPSTAT:<link_id>,<TCP/UDP>,<C/S>,<remote_ip>,<remote_port>
`
It seems that message one becomes needless. But the message one is the original one, we cannot abandon it.
And if you need to parse message one "<link_id>, CONNECT" anyway, then we think you can parse parse "<link_id>,<TCP/UDP>,<C/S>,<remote_ip>,<remote_port>,CONNECT" in the same way.

MaJerle, 2017-12-04:
I understand. I thought that if user goes from normal x,CONNECT mode to advanced one, then you will remove x,CONNECT  completelly and move to my proposal.

Anyway, if there is an upgrade with your proposal, I will be satisfied.

FayeY, 2017-12-06:
Hi @MaJerle, 
After think it twice, we are going to follow your suggestion.
We are going to add a command AT+CIPSTATUS=1/0 to enable/disable it.
If it is enabled, the message will be "+CIPSTATUS:<status>,<link_id>,<TCP/UDP>,<C/S>,<remote_ip>,<remote_port>"
And it is disabled by default, the message is the original one, "<link_id>, CONNECT"
Again, thanks for your valuable advice!

MaJerle, 2017-12-13:
@FayeY can you please comment these tickets?

https://github.com/espressif/ESP8266_NONOS_SDK/issues/65
https://github.com/espressif/ESP8266_NONOS_SDK/issues/66

closed 2018-01-11 by FayeY

MaJerle, 2018-01-16:
It is very confused how +LINK_CONN reports a new status. When connection is closed, `x,CLOSED` is received and when active, `+LINK_CONN,0,x,....` is received from ESP AT port. This is in normal operation. There is no `+LINK_CONN` message on normal connection close event.

Today, I had some network problems and my sending was unsuccessful. I received `+LINK_CONN` like below one.

From message, I can say that connection ID is `0` and first number `1` indicates `error connecting`. Does it mean that now connection is closed and I should not expect `0,CLOSED` after that?

Very confused!

```
AT+CIPSEND=0,2048
OK
> 
Recv 2048 bytes
+LINK_CONN:1,0,"TCP",1,"192.168.0.12",58678,80
SEND FAIL
```