espressif/ESP8266_NONOS_SDK #65: AT feature request: Extend AT+CIPSTART functionality

MaJerle, 2017-12-05:
In one of the issues, we found a problem with `AT+CIPSTART` command.

The root cause of the problem is that user has to specify connection number he wants to use to start connection. Everything is ok, except if you also have server mode enabled.

If you have server enabled and you to `AT+CIPSTART=0,...` but at the time of writing server populates the same connection, you have a problem because you are "confused". In this case, ESP should be the one who assings connections, either for server or client as currently we have 2 sources assigning connections (client via AT, ESP itself).

Proposed AT command which would fix this problem is:

`AT+CIPSTARTEX="TCP","ip",port\r\n` and it would return us first assigned connection number, such as `+CIPSTARTEX=0` meaning connection number `0` was assigned for new connection. In case there is no available connections in memory, ERROR is returned. If there is a connection available (in memory), simply continue with connecting and then total response would be:

In case of successful connection:
```
AT+CIPSTARTEX="TCP".....
+CIPSTARTEX=0
0,CONNECT
OK
```

In case of assigned connection but failed connecting to ip:
```
AT+CIPSTARTEX="TCP".....
+CIPSTARTEX=0
ERROR
```

In case of no memory for connection anymore:
```
AT+CIPSTARTEX="TCP".....
ERROR
```
or
```
AT+CIPSTARTEX="TCP".....
+CIPSTARTEX=-1
ERROR
```
Negative connection can mean no available connection to start.

Please comment your thoughts.
Thank you!

Arno-Nymous, 2017-12-06:
:+1: on thatone! :smiley:
I support such an approach since this feels more like the ,natural´ way of opening sockets (since the ESP provides pure TCP data over AT commands we're basically working on socket level here). So it would be great to have such a command where I can simply tell the ESP: "I want to make a new connection, so give me any free one. ANY" :smiley: Whether it be connection 0 or 4, I don't care.

FayeY, 2017-12-14:
Hi, the link ID issue has already been fixed, see #50. 
If there is a collision, the AT+CIPSTART will return Error.
So sorry that after evaluate this feature request, we decided not to add it. 
Thanks.

MaJerle, 2017-12-14:
@FayeY, this is what nakes AT unstable. Wee need single master deciding about who will get connection number

MaJerle, 2017-12-20:
@FayeY When we can expect update and extension of AT+CIPSTART?, like I proposed, AT+CIPSTARTEX would be perfect one!

closed 2018-01-11 by FayeY