espressif/ESP8266_NONOS_SDK #66: AT feature request: Allow managing of stations when in AP mode

MaJerle, 2017-12-05:
When ESP is in access point mode, we have no info about when somebody connects. There is a command to check list of connected stations, but we are not notified over AT automatically when this happens.

Please add commands for:
- Notify me when new station connects/disconnects, such as `+APSOMETING=MAC,IP,etc` with indication about connected status
- Add an option to disconnect station from AP.

Second one is more or less not required, but first one is almost mandatory to make AP useful in user app via AT command.

FayeY, 2018-01-02:
Hi,

> Notify me when new station connects/disconnects, such as +APSOMETING=MAC,IP,etc with indication about connected status

It will be added in the next release.
```
+STA_CONNECTED:<sta_mac> // a station connects to the ESP8266 softAP
+DIST_STA_IP:<sta_mac>,<sta_ip> // ESP8266 softAP distributes an IP to the station
+STA_DISCONNECTED:<sta_mac> // a station disconnects from the ESP8266 softAP
```

> Add an option to disconnect station from AP.

Sorry that the command to disconnect a station from softAP will not be added in the official AT commands. You can add it by yourself, calling an internal API `wifi_softap_deauth`.

Thanks.

MaJerle, 2018-01-02:
Hello @FayeY,

thank you for great info.

closed 2020-12-17 by MaJerle