espressif/ESP8266_NONOS_SDK #67: Any chance to have IEEE802.11s available for ESP SOCs ?

IMAN4K, 2017-12-05:
Hi All.
I was wonder if we could have the [IEEE802.11s ](https://en.wikipedia.org/wiki/IEEE_802.11s) standard implementation available for `ESP8266` chips ?
This standard is an extension to `802.11 MAC layer` and work on top of `IEEE802.11b/g/n/ac`.
This feature will give the `ESP8266` series such a great flexibility to implement the `Mesh` networks which undoubtedly is necessary in growing `Iot` !
There is an already implementation from `freeBSD` -> [ieee80211_mesh.h](https://github.com/freebsd/freebsd/blob/01e375543f2cca888435d33af45404f00296ca0c/sys/net80211/ieee80211_mesh.h), [ieee80211_mesh.c](https://github.com/freebsd/freebsd/blob/01e375543f2cca888435d33af45404f00296ca0c/sys/net80211/ieee80211_mesh.c) which you can take a look to reduce the barriers.
looking forward to your ideas.

Best regards.

FayeY, 2017-12-18:
Hi, so sorry that ESP8266 does not support that now, but ESP32 will support mesh in the future. We are focus on developing the ESP32 mesh right now, it is coming soon, maybe in Q1 of 2018.
Thanks for your interest in ESP!

IMAN4K, 2017-12-18:
@FayeY 
Great to hear about `ESP32`.
Anyway if you need any help for `ESP8266` (because of great price) i'm here to contribute.

Regards.

closed 2017-12-18 by IMAN4K