espressif/ESP8266_NONOS_SDK #70: AT feature request: Server on multiple ports

MaJerle, 2017-12-07:
Is there any plan to support server mode on multiple ports?

Since `AT+CIPSTATUS` will be extended to immediately return status with port, ip, etc from connected client/server, we have now all the info to implement server or multiple ports.

Is there any plan for that?

FayeY, 2017-12-15:
Hi, so sorry that there is no plan to support server on multiple ports recently. But you can add it by yourself if it is needed, refer to the at example we provided in the ESP8266_NONOS_SDK/examples/at. Thanks.

MaJerle, 2017-12-15:
@FayeY, purpose of AT commands is tht we dont have to program target device.

Currently it seems that Espressif does not take care of customers with AT approach. This was clearly seen when request of at+cipstart problem was thrown on the table and ignored by Espressif team.

MaJerle, 2017-12-20:
@FayeY https://github.com/espressif/ESP8266_NONOS_SDK/issues/77

closed 2018-01-11 by FayeY