espressif/ESP8266_NONOS_SDK #71: What is the correct way to update esp_init_data_default.bin OTA?

kriegste, 2017-12-08:
There is a new version of esp_init_data_default.bin

Now, there are hundreds of ESP modules in the field. How to update them? The normal user?.bin files are being updated over the air. I remember you once wrote that these are the only parts that should be updated OTA.

Well, what is the correct way to handle the new version of esp_init_data_default.bin?

FayeY, 2017-12-18:
Hi, 
yes, OTA can only upgrade userX.bin, sorry that there is no official way to upgrade esp_init_data_default.bin. 
To upgrade esp_init_data_default.bin, you need to re-write the flash area where stores the esp_init_data_default.bin by yourself. 
Sorry for the inconvenience. 
Thanks.

kriegste, 2018-10-01:
Suggestion:

Put the esp init data into an array which has to be included directly into the source of every project.
This would remove the need of an extra sector/partition and thus save us 4096-128 = 3968 bytes or 4096-128-128 = 3840 bytes in OTA update mode.

It also enables every project to conveniently update this data. Some projects rely on e.g. byte 107 being 0xFF and so on.

It would also prevent the SDK from crashing because of corrupted/missing init data. The bootloader can simply jump to the other image which has its own working init data.

kriegste, 2018-10-19:
This would also solve all issues that are caused by the init partition being mandatory like
https://github.com/espressif/ESP8266_NONOS_SDK/issues/168
and
https://github.com/espressif/ESP8266_NONOS_SDK/issues/179
and
https://github.com/espressif/ESP8266_NONOS_SDK/issues/182