espressif/ESP8266_NONOS_SDK #72: Crash deep inside SDK when switching to wifi promiscuous/monitor mode

pjsg, 2017-12-08:
Original thread: http://bbs.espressif.com/viewtopic.php?f=7&t=8485&p=18583

I have a reproducible test case. It requires the user/sniffer.c file be edited (near the bottom) to include the ssid/password of a local network. It will then crash after a couple of iterations (taking roughly 30 seconds). I'm running this on a stock Wemos D1 Mini (i.e. 8266 with 4MB flash). This is running off the stock v2.1.0 SDK.

The code is a slightly hacked up version of the 03Wifi/Sniffer_Demo code from Espressif. It does include the wifi event_handler callback from the docs. 

The attached file is a zip file of the build tree. 

The symbol just before the crash is

4010529c T wDevDisableRx

Output is 

```
mode : sta(5c:cf:7f:80:0b:01)
add if0
Sniffer testing mode ....: 2.1.0(116b762)
event   8
event   2
mode:   0   ->   3
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with Chateau Gladstone, channel 6
dhcp client start...
event   0
connect   to   ssid   Chateau Gladstone,   channel   6
ip:192.168.1.217,mask:255.255.255.0,gw:192.168.1.254
event   3
ip:192.168.1.217,mask:255.255.255.0,gw:192.168.1.254
wifi_station_scan ok
scandone
ssid Chateau Gladstone, channel 1, authmode 3, rssi -84
ssid Gladstone IPv6, channel 1, authmode 3, rssi -84
ssid Chateau Gladstone Guest, channel 1, authmode 3, rssi -83
ssid Chateau Gladstone, channel 6, authmode 3, rssi -72
ssid Chateau Gladstone, channel 2, authmode 3, rssi -64
ssid Chateau Gladstone Guest, channel 6, authmode 3, rssi -90
ssid Chateau Gladstone, channel 11, authmode 3, rssi -87
current channel1 1--------------------------------------------13581574
state: 5 -> 0 (0)
rm 0
event   8
event   1
disconnect   from   ssid   Chateau Gladstone,   reason   8
current channel 2--------------------------------------------14754846
[len] 552, rssi -59
[len] 28, rssi -86
[len] 28, rssi -85
current channel 6--------------------------------------------15755863
[len] 30, rssi -62
current channel 11--------------------------------------------16756946
[len] 28, rssi -94
[len] 28, rssi -93
[len] 28, rssi -92
Switch out of promiscuous mode
turn off sniffer:52824
event   8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with Chateau Gladstone, channel 2
dhcp client start...
event   0
connect   to   ssid   Chateau Gladstone,   channel   2
ip:192.168.1.217,mask:255.255.255.0,gw:192.168.1.254
event   3
ip:192.168.1.217,mask:255.255.255.0,gw:192.168.1.254
wifi_station_scan ok
scandone
ssid Chateau Gladstone, channel 1, authmode 3, rssi -85
ssid Gladstone IPv6, channel 1, authmode 3, rssi -85
ssid Chateau Gladstone Guest, channel 1, authmode 3, rssi -86
ssid Chateau Gladstone, channel 2, authmode 3, rssi -62
ssid Chateau Gladstone Guest, channel 6, authmode 3, rssi -88
ssid Chateau Gladstone, channel 6, authmode 3, rssi -65
ssid Chateau Gladstone, channel 11, authmode 3, rssi -89
current channel1 1--------------------------------------------30270076
state: 5 -> 0 (0)
rm 0
Fatal exception 28(LoadProhibitedCause):
epc1=0x40105300, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
```

Anything else, please just ask.

[sniffer.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1543919/sniffer.zip)

zhang-jun-hao, 2017-12-18:
Hi @pjsg 

Please call wifi_promiscuous_enable (1) after the EVENT_STAMODE_DISCONNECTED event in the wifi_handle_event_cb function.
```
	 case	EVENT_STAMODE_DISCONNECTED:
	     os_printf("disconnect	from	ssid	%s,	reason	%d\n",	
		     evt->event_info.disconnected.ssid,	
		     evt->event_info.disconnected.reason);
             //Please call wifi promiscuous enable after disconnected!!!
             wifi_set_channel(1);
             wifi_set_promiscuous_rx_cb(sniffer_wifi_promiscuous_rx);
             wifi_promiscuous_enable(1);
             os_timer_disarm(&channel_timer);
             os_timer_setfn(&channel_timer, sniffer_channel_timer_cb, NULL);
             os_timer_arm(&channel_timer, 1000, 0);
	     break;
```

pjsg, 2017-12-20:
Thank you for this -- it was very helpful. To others who may have the same issue, you don't get he STAMODE_DISCONNECTED callback if the mode is already STATION_IDLE. 

Can you add a note into the documentation to make this more obvious...

Thanks again.