espressif/ESP8266_NONOS_SDK #74: Documentation.. pretty please?

someburner, 2017-12-13:
I'm happy to see the influx of commits/fixes, but with that comes the possiblity of breaking tweaks/workarounds people have made. As someone trying to diagnose WiFi (and other) issues for the ESP8266, it would be nice to see a little more information for commits.

Some examples from recent commits:

* 099f909: Im assuing this is referring to ESP as station. "keep connection with AP" sounds like a pretty important issue? But no links to an issue or any comment other than `internal github`. 
* 477d338: I think I have seen this behavior. But the commit message is such a tease.
  - "add max scan and min scan timer" ... **to what values**? 
  - "change the scan timer set mechanism" ... **to what?** and more importantly, **from what?**
 *  d62e5a7: This is one is fairly informative. However it leaves me wondering, what was wrong before? Does this improve noise levels? Implications for FCC, etc ?

**Proposal**: Provide a summary of whatever was discussed in `internal gitlab` in the commit message.

**Bonus points**: Add a `CHANGELOG` to this repo and update that with brief summary from your gitlab.

kriegste, 2017-12-26:
Please add a declaration and documentation for the following APIs:

espconn_tcp_set_max_retran
espconn_tcp_set_max_syn
wifi_station_set_enterprise_identity
wifi_softap_deauth

kriegste, 2017-12-26:
Please add proper documentation for

wifi_set_country

Here is a list of allowed channels by country:
https://www.juniper.net/documentation/hardware/netscreen-appliances/5gtwireless/wireless_channels.pdf

Short summary:
Those countries only allow channels 1-11:
CANADA
COLOMBIA
DOMINICAN REPUBLIC
GUATEMALA
MEXICO
PANAMA
PUERTO RICO
UNITED STATES
UZBEKISTAN

All the rest allow channels 1-13.

closed 2018-04-20 by someburner