espressif/ESP8266_NONOS_SDK #75: Documentation on Websocket

moelha, 2017-12-15:
Hello, I haven't found any documentation on using the TCP socket on the ESP266 with the NON-OS SDK. Can you please provide some?

FayeY, 2017-12-18:
Hi, sorry that ESP8266_NONOS_SDK cannot use socket, it needs to use espconn APIs instead. The [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) can support socket.

moelha, 2017-12-18:
Thanks for your reply. Where can I find documentation on the socket implementation using the espconn API?

Thanks

FayeY, 2017-12-20:
See [esp8266-nonos-sample-code](https://github.com/espressif/esp8266-nonos-sample-code) or [sample codes](http://www.espressif.com/en/support/explore/sample-codes) on ESP website.
Thanks.