espressif/ESP8266_NONOS_SDK #77: AT: Cannot increase number of connections

MaJerle, 2017-12-20:
Hello Espressif team,

I was following your tutorial to setup environment for compiling NONOS_SDK. Everything worked as expected, but then I found this:
```
at_customLinkMax = 5;
at_init();
```

where setting `at_customLinkMax` sets the maximal connections used for AT. Somehow I was very sceptic this will work on compile time as there is `libat.a` already PRECOMPILED which means if `at_customLinkMax` was set to 5 on compile time of libat, no matter what I change now, it will still be 5.

Ok, let's try, maybe I'm wrong. I set this to 10, compiled everything again, upload correct file (!) and `AT+CIPSTART=9,...` returned `LINK ID ERROR` while `AT+CIPSTART=4,...` still worked as expected.

And this makes sense to me. Espressif team did you really test this behavior? Do you know what does it mean "precompiled"? If you want to have precompiled lib, then the only way to change number of connections is:
1. By going open source with libat library so that we can compile ourself
2. By using dynamic memory inside libat and then `at_init(10)` would mean 10 connections maximal
3. By using another function `at_set_conn_memory` where first parameter would be pointer to connections array and second would be length.

Current result: No matter what you set with `at_customLinkMax`, you will still have only 5 connections as number `5` was compiled to `libat.a` library.

I'm expecting detailed explanation why is this happening and why are you playing with us!

Have a nice day.

FayeY, 2017-12-20:
Hi, 
Sorry that we cannot reproduce your problem.
Please check the AT version by command `AT+GMR`, it will show the compile time. To make sure that you flashed the new AT bin which generated with configuration `at_customLinkMax=10`.
If your problem is still unsolved, please provide your test steps, we will test it exactly as your way.
Here is our test steps:
`
AT+CWSAP="APESP","88888888",5,3
 OK
AT+CIPMUX=1
 OK
AT+CIPSTART=7,"TCP","192.168.4.2",8080
7,CONNECT
 OK
 +IPD,7,4:TEST
+IPD,7,4:TEST
+IPD,7,4:TEST
`
Thanks.

MaJerle, 2017-12-20:
I uploaded a user file from `upgrade` folder which was generated on the date and time I started compilation. This looks ok.

When doing AT+GMR, I can see a new date.

I can reproduce this error many times. Test case is the same as in your case now and does not work.

FayeY, 2017-12-20:
Hi,
The user.bin we used to test is as the attachment, please have a try.
Thanks.
[user1.1024.new.2.bin.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1575029/user1.1024.new.2.bin.zip)

closed 2018-01-11 by FayeY