espressif/ESP8266_NONOS_SDK #79: espconn_secure_set_size unable to set less than 2048

valkuc, 2017-12-28:
Why it's not possible to set buffer less that 2048 bytes? If I try to set any value < 2048, espconn_secure_get_size return 2048. Is there any restriction? The document 2c-esp8266_non_os_sdk_api_reference_en.pdf says that any value in range 1-8192 is acceptable.

davydnorris, 2017-12-28:
Can I ask why you want to set it less? Are you trying to reduce memory when it's not in use?

Secure connections are very memory hungry so you usually see posts asking how to increase it!

valkuc, 2017-12-29:
I'm using very small key for my purpose - 512 bit, and I want to set ssl buffer as low as it possible but can't go less than 2048. So I guess either document must be updated that range of values is 2048-8192, either code fixed to allow values lower than 2048

davydnorris, 2017-12-29:
I am subscribed to the other thread you are replying on too, so I saw your comment there.

Even with a key that small, I think you'll still need 2048. But you could recompile the libraries to let you try less.

kriegste, 2017-12-29:
You can try and change it here. Then recompile.
https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/mbedtls/app/espconn_secure.c#L193

valkuc, 2017-12-29:
I'm actually using libssl.  Lib mbedtls require even more memory - I just can't use it because I'm "on the edge" of available RAM.

davydnorris, 2017-12-30:
You may be able to recompile mbedTLS with just the features you need and shrink it even more than the SSL lib - might be worth a try

valkuc, 2017-12-30:
Well, thanks for clues!

closed 2018-09-21 by xcguang