espressif/ESP8266_NONOS_SDK #81: AT feature request: Confirm +IPD reception

MaJerle, 2018-01-03:
Hello team,

Before I was posting this I was thinking about UART with RTC/CTS to block sending data from ESP side to host MCU, but I don't think this is the correct idea.

So, today if we connect to server, or client to us, if there are data to be read, +IPD statements are received to host MCU. In some cases, many of them.

What about sending confirmation to ESP when we want more data?
This would most probably make sense only for SSL/TCP connections which have this feature in basic protocol.

When data are received `tcp_recv` callback is called from LwIP core. In this function you have to call `tcp_recved` to notify stack that data were received and we expect more data (acknowledge). What if we add an option where user must manually send AT to confirm this? This would be handy if you receive a data packet and now you don't want to receive more until you perform some action, but still want to communicate with device (RTS/CTS is then not possible here).

FayeY, 2018-01-04:
Hi,
We will provide an API to register the `tcp_recv` callback by yourself, and also provide the API which runs like the `tcp_recved`, then you can develop your own AT command to confirm +IPD reception. Is it OK?

MaJerle, 2018-01-04:
Personally, I was thinking as an additional option with AT commands. By default, behavior is still the same (ESP sends as many packets as it has available), but if we turn it on somehow (`AT+IPD=1` ?) we would need to response to received packets.

Anyway, let's see what you will provide and then I will see on my own what i can do here. Personally, I'm more comfortable if you put it to official version so I can develop a general AT lib parser for all users.

Thanks.

FayeY, 2018-01-04:
Thanks for your precious time working on the ESP. 
I edited the comment above, not sure if you noted it. 
So I will just comment it again.
Please note that SSL connection cannot support this feature. Only normal TCP can support it.
Sorry for the inconvenience.

MaJerle, 2018-01-04:
I understand this. I think that this would be required only on TCP as TCP is "sensitive" to be precise protocol. This is not the case for UDP and SSL is not a highrunner on connection usage, so if there is only feature for TCP, it is enough.

FayeY, 2018-01-04:
We are going to add a function like `struct espconn* esp_at_get_espconn(int link_id);`.  After getting `struct espconn*`, you can regist your recv_callback by `espconn_regist_recvcb` and do something else.
Unfortunately our espconn did not make it that way, but set two different ways internally, one for normal TCP, another for SSL connection. So only normal TCP can support this feature now.
Again, sorry for the inconvenience...

FayeY, 2018-01-11:
Hi,
1. using the "at_test.zip" attached, you can define  {"+ESPCONN", 8, NULL, NULL, at_setupCmdEspconn, NULL} to add a new command.
`AT+ESPCONN=<Link_ID>,<enable_user_espconn_cb>`
[at_test.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1621021/at_test.zip)

2. We added messages for #66, and also command AT+SYSMSG for #63.
[4A-ESP8266_AT_Instruction_Set__EN.pdf](https://github.com/espressif/ESP8266_NONOS_SDK/files/1621156/4A-ESP8266_AT_Instruction_Set__EN.pdf)

Please have a try. 
Thanks.

MaJerle, 2018-01-11:
Hello @FayeY,

1. Does it mean that +IPD is not automatically sent in this case? How do I report to LwIP "I received everything", ACK remote side?

2. Sounds good. When will be this merged to master?

Best regards,
Tilen

xcguang, 2018-01-11:
Hi @MaJerle ,
1. Yes, at core will not read the data from remote TCP, so no `+IPD`. After you call espconn_recv(), lwip will update TCP window automatically.
2. it will be merged to master next release.

MaJerle, 2018-01-11:
Hello @xcguang,

1. Does it mean that this feature can be somehow used to read data "on demand". How can it be implemented that ESP only notifies user `I have data` and then user executed `Get me data, length of x bytes`.?

xcguang, 2018-01-11:
Hi @MaJerle ,
for example:
1. You regist your espconn recv_cb like `at_recv_cb(void*, char* , unsigned short len)`, when receive `AT+ESPCONN` 
2. You can print message to MCU, when running`at_recv_cb(void*, char* , unsigned short len)`,  the parameter `len` is the total length of buffered data in lwip. The message is maybe like `+DATA=len`
3. Then the mcu can send command to get data like `AT+GETDATA=len`
4. When process the command `AT+GETDATA`, you can call `espconn_recv` to read data and send data to MCU.

jasoroony, 2018-02-24:
I've just worked out how to flash my ESP8266 ESP-01 module with the latest binary supplied in this project, but I don't know how to build a new version or use these new APIs mentioned in this thread.  This library is far more stable than what originally came loaded with my board, a 2016 firmware, so thanks for continuing to improve the product.

I'm implementing an FTP server with this module, and while most things work now, the one thing that doesn't work is uploading a file to my MCU "server".  The cluster size on my SD card is 16KB so I optimise writes to that boundary as documented in the ChaNFS library, for the fastest write speed.  This introduces a large delay where the MCU can't empty the Serial input from the ESP-01 before an overrun occurs and I miss the starting +IPD signal on some packets (I see many extra characters in my debug logs, but the start of the message was lost so the software ignores them until the next +IPD comes in).

I think it would be really good if there were a new/special AT mode added that does what's mentioned in this thread, for all of us, rather than everyone compiling their own firmware version.  It doesn't have to be turned on by default, since this library has been around for such a long time and has many existing users.

If this can't be done, I've created a special FTP client that pauses TCP/IP traffic when I know I've sent enough bytes for the SD write to start (which is SD card specific, so isn't ideal at all) so I can work around it, it's just a shame I can't use my favourite FTP client at the moment for uploads.  Everything else works great though...

Thanks for your efforts.

MaJerle, 2018-09-21:
@xcguang this commands do not solve the issue. Personally, I'm not sure why, but Espressif is ignoring very good proposals to make this library 100% useful. We would need to implement reception once we have the data on device, then we can expect new data from device. Currently by manually reading the data, if connection is closed and data are in buffer, they are simply flushed out. This is major issue.

xcguang, 2018-09-21:
Hi @MaJerle ,
You are right, if  the connection is closed and the new one comes, the old data are lost.
But 
1. I am not sure if you will read the data which come from the old connection, if not, the memory will leak. 
2. If the old data is recorded after the new connection is established, maybe user will confuse which connection the data come from.
It seems another command need be added, which indicates when the old data is lost, value like 
1: discard the data when the connection is closed
2: discard the data when the new connection is established
3: do not discard the data until user read them, and the connection uses the old link id can not be established. If user never read data, the old link id will be used, and it means that you cannot create 5 connection until you read the old data.
How about this?

MaJerle, 2018-09-21:
@xcguang 

If I reply to your topics, then:

1. Imagine it is critical information (payment) and on server side payment was successful and data returned back to you as client. But since connection was closed, you missed that data and you don't know about success.
2. Instead of adding new command, why not go with my approach? The point we achieve here is that remote side won't send more data if there is no acknowledge from our side. Simple as that. So effectively we have the same as if we manually read the data, but connection won't be closed unless we read all data. If connection is closed before, we know that not all data were received back.

Please consider again in your team.

MaJerle, 2018-09-21:
Hello @xcguang,

it was very clear explanation and yes, here I agree. What we have to guarantee at the end is that even if remote has closed the connection while data are in ESP buffer waiting for user to read it manually, you should not send `x,CLOSED` event user.

It also means that even if remote side closed connection and connection is actually available to start a new one, you should prevent doing this unless there is no data in the buffer. Otherwise we can simply loose part of data which can be very critical for application.

So, at the end it would be:

1. Remote send data to ESP, ESP acknowledged
2. MCU reads data manually
3. Remote sends again data to ESP, ESP acknowledged
4. Connection closed by remote side.
    - Currently `x,CLOSED` is sent to MCU, reporting connection closed, **this is wrong**, do not send it here.
    - Data are still in internal buffer
5. MCU reads data manually
6. Send `x,CLOSED` to MCU, **this is missing**

If there is `AT+CIPSTART=x,...` command between points `4` and `6`, ESP should return `ERROR` reporting that connection is still virtually active as we have data to read. The same applies if server wants to use the same connection again, it should not use it until we read data.

Of course this could mean that if user forgets to read data, connection becomes inaccessible forever. Here I would add default timeout when `x,CLOSED` is reported to user and data are deleted if there is no manual read.

Do you understand the point? At the end of the day, we have to guarantee that `x,CLOSED` is sent *LAST*, at the end of everything and that we know after that message, story with this connection is finished, no more usage. Currently we can still read some data after we receive `x,CLOSED`. This does not make sense.

Waiting for your valuable reply. Please consider this with your team.

dsvd1979, 2019-06-07:
Hi @MaJerle 

Im also working on ESP8266 1MB (512kbx512kb) with 8MB of external flash. Have you ever resolved the issue about the +IPD being received randomly from ESP01. This interferes with my current setup as I am doing TCP writes to ESP with AT command and expect certain return, but inbetween I receive +IPD command from ESP to MCU with new TCP data. This the write to TCP fails as it receives wring reply from ESP.

Regards

MaJerle, 2019-06-07:
In lib I have developed on my own, I'm not using TCPBUFF option but only AT+CIPSEND to immeduately send the data. I keep "active command" until I get response either SEND OK or SEND ERROR.

If I receive +IPD in-between, this is not a problem for me. i have also discovered that manual TCP read is much more stupid than automatic push from ESP. Imagine that you get notification about data available and immediately conn CLOSE event. Then if you are server and a new client starts connection, your data from previous connection are lost.

closed 2019-06-07 by MaJerle

dsvd1979, 2019-06-10:
Hi @MaJerle 
Thanks for the feedback. Do you have an example of what you have done?

My MCU is actually a arduino UNO trying to communicate to the ESP via UART.

Thanks in advance.

Regards
Dirk

MaJerle, 2019-06-10:
> Hi @MaJerle
> Thanks for the feedback. Do you have an example of what you have done?
> 
> My MCU is actually a arduino UNO trying to communicate to the ESP via UART.
> 
> Thanks in advance.
> 
> Regards
> Dirk

Everything I have is available here:
- https://github.com/MaJerle/ESP_AT_Lib_res (examples, etc)
- https://github.com/MaJerle/ESP_AT_Lib (lib itself)