espressif/ESP8266_NONOS_SDK #84: smartconfig broken

someburner, 2018-01-11:
Linking against this SDK at master, I get an error on boot for smartconfig:

```
	 ---> SDK: 2.2.0-dev(a797448) <---

initSmartConfig = true

bcn 0
add ifstart smart (version V2.5.4)
SC version: V2.5.4
err1,exceed max time value
Wifi OP mode: 2 -> 1
```

I tried all kinds of stuff- delaying smartconfig init, various wifi settings, calling `esptouch_set_timeout` etc, none of that works. However, linking against SDK v2.1.0 libsmartconfig.a, the problem disappears and code works like before.

 I believe it is the same issue as [the RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK/commit/fe558a98255e01cb1df0a2d3e45e760689acdef6) perhaps?

xuhongv, 2018-01-13:
@someburner 

- I also find the same issue , error message "err3,exceed max time value" ,can you tell me how do you to solve , please ?

******************************************************************************************
[image]

someburner, 2018-01-14:
Not sure about on windows but just download this same repo at the [version 2.1.0](https://github.com/espressif/ESP8266_NONOS_SDK/tree/v2.1.0) tag and overwrite libsmartconfig.a with the old one.

FayeY, 2018-01-15:
Hi, so sorry for the inconvenience.
We just updated the SDK, which should solve this issue, please have a try.
Thanks.

someburner, 2018-01-15:
@FayeY thanks- it's all good to go again now.

closed 2018-01-15 by someburner