espressif/ESP8266_NONOS_SDK #86: when use smartConfig , warning message  "exceed max time value" !

xuhongv, 2018-01-13:
- 我尝试用您们这个NONOSSDK开发，在MQTT的sample之上，加上smartConfig配网，发现报错截图如下，请问，这是啥原因 ！：
 
  - 环境是安信可集成的。

[image]

FayeY, 2018-01-15:
你好，
请问，SDK 的版本号是多少？能否使用最新的 SDK 看是否还有问题？
如果问题仍然存在，麻烦提供你的测试程序，我们将协助进行查证。
谢谢！

xuhongv, 2018-01-15:
@FayeY  SDK版本是最新的：2.2.0-dev(a797448) 。demo的开发环境为安信可一体化环境。

虽然我个人已经解决，方法是把版本退回到2.1.0 ，但是mqtt连接不稳定。

希望完善之后，可以提醒我一下，谢谢。

[ESP8266_NONOS_SDK-master.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/1630097/ESP8266_NONOS_SDK-master.zip)

FayeY, 2018-01-15:
你好，
我们刚刚更新了 ESP8266 non OS SDK，解决了这个问题，麻烦进行测试，如果仍然有问题，可以再告知。
非常抱歉给你带来不便。
请知悉。谢谢！

closed 2018-01-16 by FayeY