espressif/ESP8266_NONOS_SDK #87: system_adc_read_fast() always returns 1024 as output

ashish28ranjan, 2018-01-14:
### Basic Infos

#### Hardware
Hardware:	NodeMCU v3 (ESP-12E)
esp8266/Arduino core Version:  	2.4.0-rc2
System SDK Version:  	2.1.0(deb1901)
Arduino IDE:  1.8.5

### Description

I need to use `system_adc_read_fast (uint16 *adc_addr, uint16 adc_num, uint8 adc_clk_div)`
to get sampling rate close to 100ksps. And I am able to achieve that using `adc_clk_div = 32`.

However, when I print out the values stored in `adc_addr`, **I always get 1024**. Why is this?

I have to sample audio from an electret microphone. This is the schematic:

[image]

### Observations
1. It returns correct values when I use `system_adc_read()` (except the sampling rate is then 10ksps which is not desirable). Hence, I'm sure it's not a problem of the above schematic.
2. **I get 1024 even when I connect A0 to GND.**

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB (1MB SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?qio? (don't know!)
Flash Frequency:	?40Mhz? (don't know!)
Upload Using:		AVRISP mkII (serial)
Reset Method:		nodemcu

### Sketch

```cpp

// Expose Espressif SDK functionality - wrapped in ifdef so that it still
// compiles on other platforms
#ifdef ESP8266
    extern "C" {
        #include "user_interface.h"
    }
#endif

#define num_samples 512
uint16_t adc_addr[num_samples]; // point to the address of ADC continuously fast sampling output
uint16_t adc_num = num_samples; // sampling number of ADC continuously fast sampling, range [1, 65535]
uint8_t adc_clk_div = 32; // ADC working clock = 80M/adc_clk_div, range [8, 32], the recommended value is 8

int i = 0;
unsigned long start = 0;
unsigned long total = 0;
unsigned long tim = 0;

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
}

void loop() {
    // put your main code here, to run repeatedly:
    #ifdef ESP8266
        // Serial.println(system_get_sdk_version());

        wifi_set_opmode(NULL_MODE);
        system_soft_wdt_stop();
        ets_intr_lock( ); //close interrupt
        noInterrupts();

        start = micros();

        // Serial.println(system_adc_read());
        system_adc_read_fast(adc_addr, adc_num, adc_clk_div);

        unsigned int tot = micros() - start;

        interrupts();
        ets_intr_unlock(); //open interrupt
        system_soft_wdt_restart();

        tim += tot;
        total += num_samples * 1000000.0 / tot;
        i++;

        for (int j=0; j<adc_num;  j++) {
            Serial.println(adc_addr[j]);
        }

        if (i == 100) {
            // Serial.print("Sampling rate: ");
            // Serial.println(total / 100);
            // Serial.print("It lasted: ");
            // Serial.println(tim / 100);

            i = 0;
            tim = 0;
            total = 0;
        }
    #endif
}
```

### Debug Messages
Response from ESP:
```
1024
1024
1024
1024
1024
1024
1024
```

Duiesel, 2023-05-07:
Any updates on it?

TheCrazyT, 2025-01-12:
Still no solution for this?
All I found was that you probably need to change vdd33_const.
There is a reference in the "core_esp8266_phy" but it is kinda ugly if I change it there directly. (and I'm not shure if it would have any impact)

kriegste, 2025-01-12:
Max. input voltage for the analog pin is 1V.

TheCrazyT, 2025-01-13:
> Max. input voltage for the analog pin is 1V.

that wasn't the problem for me.
i'm now using a 160mhz cpu setting with adc_clk_div=16, since 32 seem to output a constant value.