espressif/ESP8266_NONOS_SDK #88: Uart flooded with "sta trace 0"

kriegste, 2018-01-15:
After the last update from this morning, my uart log is flooded with

sta trace 0
sta trace 0
sta trace 0
sta trace 0
sta trace 0
sta trace 0
...

Why is that and what does it mean?

valkuc, 2018-01-15:
Same for me but only when using system_phy_freq_trace_enable(1) in rf pre init function.

kriegste, 2018-01-15:
Strangely, I have a second device that does not show this behavior. Same firmware, same WiFi. I rebooted and power cycled both of them several times. Device 1 floods the log, device 2 does not.

FayeY, 2018-01-25:
Hi, is there any reason that make you use system_phy_freq_trace_enable(1)? 
The "sta trace 0" means that frequency trace function is working, it means that the RF frequency offset of your board is too large. In this case, it is suggested that you should check the hardware of this specific board for its abnormal RF frequency offset.
Thanks.

valkuc, 2018-01-25:
Ocasionally ESP8266 stop connecting to access point displaying that AP not found, reconnecting 1s. In my case system_phy_freq_trace_enable(1) fix that issue for some time but then issue occured again. I can't figure out cause of this because it happens too rare. Currently I'm not using freq trace.

kriegste, 2018-01-25:
In my case the affected hardware is a standard ESP-01 module. In the mean time I found several other modules with the same "problem". I doubt the hardware is faulty since they work perfectly otherwise and there was no flooding with earlier versions of the SDK.
I wanted to have all calibration functions enabled just to be sure.

I disabled system_phy_freq_trace_enable for now.

kriegste, 2018-04-13:
Maybe Espressif can explain in detail what system_phy_freq_trace_enable does. My interpretation is that it is some kind of calibration routine. But there might be a misunderstanding.

FayeY, 2018-04-16:
system_phy_freq_trace_enable is to enable RF frequency offset trace. 
If your module will work in the high-temperature environments, you need to call it.
If not, then this API should not be called.
And if your module works in normal temperature, and the API was called, and "sta trace 0" shows, it means that your module has a hardware issue, so the software needs to do the RF frequency offset trace. In this case, you need to revise your hardware.

closed 2018-04-16 by FayeY

kriegste, 2018-05-02:
Some observations on the ESP-01 module that shows the error message:

If I heat the module to ~100°C, the message no longer appears. The module continues to work.
If I cool the module to ~0°C the message no longer appears. The module continues to work.
If I cool down the module further (~-30°C) the module gets into a boot loop (until it warms up again):

Fatal exception 3(LoadStoreErrorCause):
epc1=0x402b68e2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4000e2a4, depc=0x00000000

0x402b68e2 seems to be in a function called "phy_dig_spur_set".

This is no question or error report, just an observation. Maybe it is helpful for Espressif anyway.