espressif/ESP8266_NONOS_SDK #89: AT random resets

MaJerle, 2018-01-18:
I'm using latest AT software from this repository, compiled on my own.

After few hours, device resets:

```
AT+CIPSEND=19,4<\r><\n>
<\r><\n>
OK<\r><\n>
> <\r><\n>
Recv 4 bytes<\r><\n>
<\r><\n>
SEND OK<\r><\n>
<\r><\n>
+IPD,19,4,37.187.106.16,1883:p<2>%?rl<\0>l??|<\0>?l?|<2><\f><\f><\f>?<\f>l?<\f>b|??<2>?<18>?r?b?<\f>b??nn?lnn???<\f>b<28>p??lrlrlp?n?<16><2><\f><\f>?<\f>l<\f><\f><\f><\f><\f><\f>b<\f>n?|<2>?l?|?<\f>b??nn?<\0>l??l`<2>?<18><18>nn<\f>l`<2><14><2>nr???n<\f><\f>?rl`<2>p?n?<16><2><\f><\f>r?????<\f><\f><\f><\f>b<\f>n?|<2>????b??nn?<\0><\f><\f>l`<2>?<18><18>nn<\f>?<14>l<14><2>nr???n<\f><\f><18>?l`<2>p?n?<16><2><\f><\f>r????????pp<\f><\f>n?|<2>????b??nn?<\0><\f>?l`<2>?<18><18>nn<\f>l`<2><14><2>nr???n<\f><\f>b?l`<2><14>r??n<\f><\f>b?l`<2>rl??n???nn?????r<18>nnn?<\f>l?b?br<16>??????lrl?<2>?l?<18>???<2>?<\f><\f>?lll~rl?<2>?llln??<\0>?<\f>l?ll<14>l?<2>?l??n??<2>n??n~<18><2>??ll??b<\f><\f>llpb?<2>br???l`nl?p??l`bl`<2>l??n<2><\f>?n???nl?|<18>b<18><\f><\f><2>l<\f><\f><\f>l`<2>rl??<2>?<18>?n??r??n|?<\f>?l?prl<28>b?<18><2><\f>?|<2>r?<2>l?n?<\f>?n?<\0>ll`<2>??r?l?l<18>?<\f><\f>?l`<2>??r?l?l<18>?<\f><\f><\f>l`<2>??r?l<\f>??<\0>?<\f><\f>ll`<2>rl??rl????b<\f>??b<28>|<\f>lb???<2>b??l????lb??n??<18>nn?<16><2><\f><2><\f>??<14><2>l?<\f>?<\f>l??l??<30>?l?l?<28>rl?<2>n?<\0>????b<\f>ll???<18><\f>b<\f><2><2>????b<\f>l<\f>?b<28>rlrl<\r><\n>
ready<\r><\n>

```

Why?

FayeY, 2018-01-22:
Hi, sorry that we cannot reproduce your problem.
Could you provide more details about your test steps? We will try to test it exactly as your way.
And could you provide the UART1 log for debugging?
Thanks.

MaJerle, 2018-01-23:
I have nothing special, bunch of `AT+CIPSEND` commands and `+IPD` received data from MQTT server. I'm testing my MQTT application and I'm sending data as soon I receive a confirmations from MQTT broker. Communication is always active, random resets. Is it possible the problem is not software but hardware?

FayeY, 2018-04-16:
Closing as can't reproduce. Please reopen if the issue persists and you can supply more details, such as the UART1 log, for debugging.

closed 2018-04-16 by FayeY