espressif/ESP8266_NONOS_SDK #94: SoftAP extra beacon

devyte, 2018-01-29:
Hello,

this issue is meant to cover what is described [here](https://bbs.espressif.com/viewtopic.php?f=66&t=3328&p=11258).

In summary, the SoftAP broadcasts Channel Switch Announcements (defined in 802.11h) when the STA connects while in STATIONAP mode and the STA channel is different than the already-running SoftAP channel. However, there seems to be an extra beacon broadcast, which can cause the devices connected to the SoftAP to not follow at the correct time.

For completeness, this is copied from the above link:
```
SN      SSID           CurrChan   CSA
=====   ============   ========   ========
1058    ESP_1BEB89     9
1062    ESP_1BEB89     9          Mode: 1, Number: 4, Count: 3
1064    ESP_1BEB89     9          Mode: 1, Number: 4, Count: 2
1066    ESP_1BEB89     9          Mode: 1, Number: 4, Count: 1
1068    ESP_1BEB89     9
(then no more beacons because the switch occurred)
```

I suspect this is a counter in the code that is either in the wrong place, or off by 1, and is likely an easy fix.

artua, 2018-03-09:
The same problems with iOS

devyte, 2018-04-20:
@FayeY do you know if this has been looked at? It has been 3 months since I opened the issue, and longer since it was posted in your bbs forum.

devyte, 2019-03-13:
@FayeY has there been any progress here? It has been over a year since this was opened. The fix should not be difficult, I suspect.

FayeY, 2019-03-13:
Hi, will the devices connected to the SoftAP follow to the new channel but a little later; or the devices will stay at the old channel, do not follow to the new channel?

devyte, 2019-03-13:
I think the devices connected to the softap switch channel at the correct time, but then they don't find the ap in the new channel, because the softap stays on the old channel for an extra beacon before switching.

devyte, 2019-11-05:
Ping?