espressif/ESP8266_NONOS_SDK #98: during erase flash, send data to Serial。module will wdt reset。

chenkaiyao, 2018-02-07:
### Sketch

```cpp
extern "C" {
#include "c_types.h"
#include "ets_sys.h"
#include "os_type.h"
#include "osapi.h"
#include "spi_flash.h"
}

extern "C" uint32_t _SPIFFS_end;

void setup()
{
  Serial.begin(115200);
}

uint32_t _sector = ((((uint32_t)&_SPIFFS_end - 0x40200000) / SPI_FLASH_SEC_SIZE));

void loop()
{
  while(Serial.available()>0)
  {
    Serial.print(Serial.read());
  }
  
  Serial.print("erase\r\n");
  noInterrupts();
  spi_flash_erase_sector(_sector);
  interrupts();
}
```

### Debug Messages

```
erase
erase

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v074402e2
~ld
erase
```

kriegste, 2018-02-07:
What does noInterrupts do? It is not a NONOS api. Also please do not use loops that might not be exited. Use timers instead.

valkuc, 2018-02-07:
This is Arduino framework for ESP8266 NONOS SDK. You probably need to ask here https://github.com/esp8266/Arduino

Alexis-Polegato, 2018-03-05:
I'm having the same issue ; when using `spi_flash_erase_sector` the module may reset with the following result:  

> Fatal exception 0(IllegalInstructionCause):
> epc1=0x4020a86c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

eriksl, 2018-09-07:
Probably a UART interrupt handler in FLASH (not IRAM). During flashing, IRAM cache is unmapped, UART interrupt -> crash. Never place interrupt handlers in FLASH.

xcguang, 2018-09-21:
@Alexis-Polegato ,
Could you provide us your demo to reproduce it?

closed 2019-03-15 by xcguang