espressif/ESP8266_NONOS_SDK #100: MDNS Responder only responds to the first request.

IanAber, 2018-02-28:
When the mDNS responder gets the first request the system correctly returns the DNS info to the caller but then it stops responding.

devanlai, 2018-05-08:
To add more information, I have observed that:

1. The responder announces itself with all records when first enabled as expected
2. The responder never directly responds to A record queries for the mDNS hostname
3. The responder does respond to service discovery queries and service record queries and includes the A record in the additional records section when responding to a service record query as recommended by the RFC.

Boodorgad, 2019-02-25:
I observed the same as devanlai reported.

And:
The Response works only if you query exactly during the ESP8266 announces itself. On any other query you won't get a response. (So it does not really work but it's just a coincidence)

Tested Version:
AT:1.7.0.0(Aug 16 2018 00:57:04)
SDK:3.0.0(e27cf60)

devanlai, 2019-02-25:
Assuming the source code in the SDK is used for the AT firmware, there are a number of deficiencies that match what I've observed:

- [It only handles two types of mDNS messages](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/lwip/core/mdns.c#L875) - queries for all local service types and queries for the registered service.
  - It does not process queries for A records, ever.
  - It does not parse the known-answers section to determine if it is redundant to respond.
- [It does not check for name collisions](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/lwip/core/mdns.c#L1031) - it just sends out four initial announcement packets.