espressif/ESP8266_NONOS_SDK #101: Custom handling of SmartConfig Acknowledgement

slaff, 2018-03-01:
In my project I am using SmartConfig and sending the SSID and encoded data in the PASSWORD field. As far as I can see the current NONOS SDK tries itself to connect to the SSID using the PASSWORD and once connected sends back Acknowledgement (ACK) to the cellphone. In `esp-idf` the acknowledgement can be handled in the user code as seen in [smartconfig.c](https://github.com/espressif/esp-idf/blob/master/components/esp32/smartconfig.c). I tried to reproduce the same thing in NONOS but I am missing one critical piece of information. The data that needs to be sent has the following structure:
```
    struct {
        uint8_t token;            /*!< Smartconfig token to be sent */
        uint8_t mac[6];           /*!< MAC address of station */
        uint8_t ip[4];            /*!< IP address of cellphone */
    } ctx;
```
I can calculate the token and get the mac of the station but I do not know how to get the ip address of the cell phone that initiated the smartconfig using EspTouch.

Do you have any hints how to get it or can you change the SDK to allow the ACK to be send from the user code as in `esp-idf` ?

TimXia, 2018-07-10:
@slaff In NONOS SDK, the structure is in libsmartconfig.a. In esp-idf, it is open-sourcing. Both NONOS SDK and esp-idf will allow the ACK to be sent.

TimXia, 2018-07-25:
@slaff Sorry it's not open sourced in NONOS SDK.