espressif/ESP8266_NONOS_SDK #103: Safe defaults for scan threshold and dwell time

devsaurus, 2018-03-01:
The structs `station_config` and `scan_config` in `include/user_interface.h` were augmented with new members in https://github.com/espressif/ESP8266_NONOS_SDK/commit/b4478720e5f12b78438dc8bb8d32b482b74734de#diff-3d8510b5f68b1ce4e40ea4515fff75e5

Is it sufficient to initialize with all-zero or do we have to set specific values in order to retain equivalent functionality compared to SDK 2.1.0? Please provide recommended default values for them.

ustccw, 2018-04-20:
@devsaurus 
for `struct station_config`, you could set thresholds for signal and authmode, it will ignore the AP which below the thresholds. you could set `threshold.rssi = -127` and `threshold.authmode = AUTH_OPEN`

ustccw, 2018-04-20:
@devsaurus 
for `scan_config`, the recommended default values  is up to  what you need. To do a balance between `accuray` and `total scan time`. such as in a strong network, you could set `scan_time.active.min=100` && `scan_time.active_max=300` for active scan, `scan_time.passive=100` for passive can.

devsaurus, 2018-04-20:
Thanks for the recommendations @ustccw!

Regarding `scan_config`: We will probably expose this API on Lua level for the nodemcu firmware in the future so the user can balance based on his application scenario. Do I understand right that the settings
```c
scan_time.active.min=100
scan_time.active.max=300
scan_time.passive=100
```
reproduce the scan configuration for SDK 2.1.0, which didn't provide this config capability?

devyte, 2018-04-20:
I apologize for jumping in, but I have a question on this same topic.
What happens if the following is set:
```
scan_time.active.min=0;
scan_time.active.max=0;
scan_time.passive=0;
```
?

kriegste, 2018-04-20:
This is how I interpret the original question:

What were the defaults in earlier SDK versions, when we did not have these options?

ustccw, 2018-04-23:
@devsaurus @devyte @kriegste Hi, the following is a set of test reference data, testing under the weak network.

1) passive scan in networking mode(ESP8266 got ip):  
[image]

2) passive scan in non-networking mode(ESP8266 without ip):
[image]

3) active scan in networking mode:
[image]

4) active scan in non-networking mode:
[image]

ustccw, 2018-04-23:
@kriegste the defaults in earlier SDK versions:
passive: 120 ms
active.min: 60 ms
active.max: 120ms

ustccw, 2018-04-23:
@devsaurus yes, and it is up to your networking.

devyte, 2018-04-23:
@ustccw thank you very much for this information! It is a big help for understanding the scan times!

devsaurus, 2018-04-23:
@ustccw thank you very much for the valuable information!

closed 2018-04-23 by devsaurus