espressif/ESP8266_NONOS_SDK #109: Station can't connect to AP at channel 13 with hidden SSID

devsaurus, 2018-03-27:
One of my test setups coincidentally consists of an access point that is configured to channel 13 (country DE) and hides its SSID "guest". With SDK 2.2.0 my code is not able to connect as station to this AP.

Connection can be established if I either
* configure AP to channel <13, e.g. channel 1 (and keep SSID hidden)
* enable SSID boadcast on AP (and keep channel 13)

Meaning that connection is **not** possible only when AP is at channel 13 **and** SSID is hidden.

The code as it is works and connects to my AP at channel 13 with hidden SSID:
```
add if0
station config set
connection initiated
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with guest, channel 13
dhcp client start...
ip:<ip>,mask:255.255.255.0,gw:<gw>
```

Connection can't be established if I comment the call to `wifi_set_country(&wifi_country);` or change `.policy=WIFI_COUNTRY_POLICY_MANUAL` to `.policy=WIFI_COUNTRY_POLICY_AUTO` (which is the default setting without `wifi_set_country()`):
```
add if0
station config set
connection initiated
scandone
no guest found, reconnect after 1s
reconnect
scandone
no guest found, reconnect after 1s
reconnect
scandone
no guest found, reconnect after 1s
reconnect
...
```

It seems that it's required to set the new wifi_country API to support such APs. Is this expected behavior? 

The code below is a simple test case and should compile like the other SDK examples.
```
$ make COMPILE=gcc BOOT=none APP=0 SPI_SPEED=40 SPI_MODE=QIO SPI_SIZE_MAP=0
$ esptool.py --baud 460800 write_flash 0x00000 ../bin/eagle.flash.bin 0x10000 ../bin/eagle.irom0text.bin 0x3fc000 ../bin/esp_init_data_default_v08.bin 0x3fe000 ../bin/blank.bin
```
```c
/*
 * ESPRESSIF MIT License
 *
 * Copyright (c) 2016 <ESPRESSIF SYSTEMS (SHANGHAI) PTE LTD>
 *
 * Permission is hereby granted for use on ESPRESSIF SYSTEMS ESP8266 only, in which case,
 * it is free of charge, to any person obtaining a copy of this software and associated
 * documentation files (the "Software"), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the Software is furnished
 * to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all copies or
 * substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */

#include "osapi.h"
#include "user_interface.h"

/******************************************************************************
 * FunctionName : user_rf_cal_sector_set
 * Description  : SDK just reversed 4 sectors, used for rf init data and paramters.
 *                We add this function to force users to set rf cal sector, since
 *                we don't know which sector is free in user's application.
 *                sector map for last several sectors : ABCCC
 *                A : rf cal
 *                B : rf init data
 *                C : sdk parameters
 * Parameters   : none
 * Returns      : rf cal sector
*******************************************************************************/
uint32 ICACHE_FLASH_ATTR
user_rf_cal_sector_set(void)
{
    enum flash_size_map size_map = system_get_flash_size_map();
    uint32 rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;

        case FLASH_SIZE_64M_MAP_1024_1024:
            rf_cal_sec = 2048 - 5;
            break;
        case FLASH_SIZE_128M_MAP_1024_1024:
            rf_cal_sec = 4096 - 5;
            break;
        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}

void ICACHE_FLASH_ATTR
user_rf_pre_init(void)
{
}

void ICACHE_FLASH_ATTR
init_done(void)
{
	wifi_set_opmode(STATION_MODE);
        struct station_config cfg;
        os_memset(&cfg, 0, sizeof(cfg));
        os_memcpy(cfg.ssid, "guest", 5+1);
        os_memcpy(cfg.password, "pwd", 3+1);
        wifi_station_set_config_current(&cfg);

        wifi_country_t wifi_country = {.cc="CN", .schan=1, .nchan=13, .policy=WIFI_COUNTRY_POLICY_MANUAL};
        wifi_set_country(&wifi_country);

        os_printf("station config set\n");
        if (wifi_station_connect())
          os_printf("connection initiated\n");
        else
          os_printf("failure from wifi_station_connect()\n");
}

#include "driver/uart.h"

void ICACHE_FLASH_ATTR
user_init(void)
{
  uart_init(BIT_RATE_115200, BIT_RATE_115200);
	system_init_done_cb(init_done);
}
```

ustccw, 2018-05-07:
@devsaurus 
Hi, this is the expected behavior when set AP to channel 13, and hidden it.
1) if set `.policy=WIFI_COUNTRY_POLICY_AUTO`
ESP8266 would adopt default configuration: send probe request from channel 1 to channel 11 (active scan), wait valid beacon frame from channel 12 to channel 14(passive scan).  if set *hidden* AP on channel 13, the ESP8266 would not fetch AP info from beacon frame so that cannot connect to AP.

2) if set `.policy=WIFI_COUNTRY_POLICY_MANUAL`
ESP8266 would adopt user configuration, such as set `wifi_country_t wifi_country = {.cc="CN", .schan=1, .nchan=13, .policy=WIFI_COUNTRY_POLICY_MANUAL};`
ESP8266 would send probe request from channel 1 to channel 13.  if set *hidden* AP on channel 13, ESP8266 still can receive probe response, and complete the connect workflow.

CW

devsaurus, 2018-05-09:
Thanks for the explanation @ustccw!
Seems I was missing the aspect that `WIFI_COUNTRY_POLICY_AUTO` just passively scans channels >=12 even though `.cc="CN", .schan=1, .nchan=13` would suggest that the upper channels are ok for active scan.

ustccw, 2018-05-10:
thanks for your quick feedback @devsaurus !

we will close this issue but please feel free to reopen it if you would like to follow up on the question.

closed 2018-09-21 by xcguang