espressif/ESP8266_NONOS_SDK #110: ESP8266 can't connect to eduroam

Imads608, 2018-03-28:
So I have been trying different ways to connect my ESP8266 to eduroam. 
I have latest version of the sdk and have found that you might have to edit the libwpa2.a file specifically "anonymous@espressif.com" to your identity to make it work. But apparently the latest release covers this with wifi_station_set_enterprise_identity?

So the first thing I tried was using that function and setting my identity to my username
But that did not work and have been getting the below output: 
`WPA2 ENTERPRISE VERSION: [v2.0] enable
mode : sta(60:01:94:1f:d7:1a)
add if0
[mode: 0 -> 3]
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
pm open,type:2 0
state: 5 -> 2 (7c0)
rm 0
pm close 7
[disconnect from ssid eduroam, reason 204]
reconnect`

Next thing I tried was setting the identity manually itself in the libwpa2.a file to "username@example.edu" with correct padding.
Still gives me an error but for a different reason:
`WPA2 ENTERPRISE VERSION: [v2.0] enable
mode : sta(60:01:94:1f:d7:1a)
add if0
[mode: 0 -> 3]
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
pm open,type:2 0
state: 5 -> 2 (7c0)
rm 0
pm close 7
[disconnect from ssid eduroam, reason 7]
reconnect
state: 2 -> 0 (0)
scandone`

Apparently this error according to the SDK is REASON_NOT_ASSOCED (I don't even know what that refers to).

I've been stuck on this for a while so any help will be much appreciated!

Junhao-Espressif, 2018-04-23:
Hi @Imads608 ,

Could you try [wpa2_example](https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise) first?

Could you give me your code?

Thanks!

kapyaar, 2018-05-11:
/@Junhao-Espressif Can you please look at my post related to this

https://github.com/esp8266/Arduino/issues/4698

i looked at https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise and made my code to match this as well, but still no success.

/any help would be much appreciated

victorclaessen, 2018-05-18:
You may be interested in this:

https://bbs.espressif.com/viewtopic.php?f=66&t=5962