espressif/ESP8266_NONOS_SDK #113: SoftAp does not work with AT module enabled.

ThomArmax, 2018-04-04:
Hello,
I am using the NONOS_SDK 2.1.0 with ESP-Launcher dev board.
I try to create an access point and enable the AT module without success : my phone is not able to connect to the ESP8266.
```
int ICACHE_FLASH_ATTR user_init()
{
    at_init();
    
    if (!wifi_set_opmode(SOFTAP_MODE)) {
        os_printf("wifi_set_opmode failed\r\n");
        return -1;
    }

    struct softap_config config;
    os_memset(&config, 0, sizeof(config));

    os_strcpy(&config.ssid[0], AP_SSID);
    config.ssid_len = os_strlen(AP_SSID);
    config.channel = 6;
    config.authmode = AUTH_OPEN;
    config.beacon_interval = 100;
    config.max_connection = 4;

    if (!wifi_softap_set_config(&config)) {
        os_printf("wifi_softap_set_config failed\r\n");
        return -1;
    }
    os_printf("AP configured\r\n");
    
    return 0;
}
```

Debug output from UART1 (debug only) : 
```
AP configured
mode : softAP(2e:3a:e8:22:04:4d)                                    
add if1                                                           
bcn 100                                                              
add 1                                                                  
aid 1              
station: 04:d6:aa:11:9c:ee join, AID = 1
station: 04:d6:aa:11:9c:ee leave, AID = 1                              
rm 1                                                                 
add 1                                                                  
aid 1              
station: 04:d6:aa:11:9c:ee join, AID = 1
station: 04:d6:aa:11:9c:ee leave, AID = 1                              
rm 1
```

If I comment the `at_init();` line, I am able to connect to the ESP8266 access point.

Do you have an idea of what can possibly be wrong ?

FayeY, 2018-04-13:
Please try `AT+RESTORE` to initialize it first, and then test it again.

ThomArmax, 2018-04-17:
Thanks @FayeY , it seams `AT+RESTORE` solved the issue :)

closed 2018-04-20 by ThomArmax