espressif/ESP8266_NONOS_SDK #120: Do we need to blank the rf calibration data sector? Does system_restore() blank this sector?

kriegste, 2018-04-27:
Do we need to manually blank (all 0xFF) the rf calibration data sector (the one we return in user_rf_cal_sector_set)? I mean, on initial flashing/update from an (early) SDK.

Does system_restore() blank this sector? Is it enough to call this api or is it better to erase the rf cal sector manually using spi_flash_erase_sector (when updating OTA)?

Thank you very much for clarifying!

kriegste, 2018-05-05:
Observations (on a 1 MiB flash):

Only sectors 0xFD and 0xFE get blanked by system_restore(). Sort of, i.e. the first two bytes still contain something, but the rest is gone.

Sector 0xFF contains the active user bin number. So it should not be messed with except on initial flash. What other data is stored there, I have no clue.

How the SDK reacts to garbage in the rf cal sector, I still do not know either.

kriegste, 2018-05-05:
If someone wonders why I ask all that:

For testing purposes I need to copy a flash image onto a bunch of modules and I need them to work right away, connect to my WiFi and be accessible. I can not have any WiFi names/passwords in the code, though. So I use an image that contains my WiFi data in 0xFD/0xFE and after testing I use system_restore() to wipe it.
Now, of course, if there is any rf calibration data or other data specific for a chip in particular, this would get copied as well. So I need to find out what can/must be blanked and what not.

FayeY, 2018-05-10:
You can refer to the [API Guide](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf), it lists the configurations that system_restore will reset.

kriegste, 2018-05-10:
Hi, thanks. I know that list. But that begs the question what else is stored that doesn't get reset.