espressif/ESP8266_NONOS_SDK #121: err already associed! station: 9c:99:a0:xx:xx:xx leave, AID = 1

456vv, 2018-05-06:
problem:
After the "webserver_listen“ and ”webserver_discon“, the station is connected to the route and the softAP fails to work.

error:
err already associed!
station: 9c:99:a0:xx:xx:xx leave, AID = 1

tracking information:
Wait for the update sta_conf or ap_conf information.
call response_send
call data_send
call restart_10ms_cb
call webserver_listen
call webserver_discon
webserver's 192.168.4.2:33206 disconnect
scandone
switch to channel 13
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with xxxxxx, channel 13
dhcp client start...
ip:192.168.x.132,mask:255.255.255.0,gw:192.168.x.1
call webserver_discon
webserver's 192.168.4.3:46161 disconnect
err already associed!
station: 9c:99:a0:xx:xx:xx leave, AID = 1
rm 1
add 1
aid 1
station: 9c:99:a0:xx:xx:xx leave, AID = 1

kriegste, 2018-05-06:
I, too, get "err already associed!" from time to time, but I have not figured out yet which API causes it.

456vv, 2018-05-06:
problem:
Endless repeating "no XXX found, reconnect after 1s"

tracking information:
mode : sta(a0:20:xx:xx:xx:xx) + softAP(a2:20:xx:xx:xx:xx)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
call user_esp_platform_check_ip
scandone
no XXX found, reconnect after 1s
reconnect
scandone
no XXX found, reconnect after 1s
reconnect
call user_esp_platform_check_ip
scandone
no XXX found, reconnect after 1s
reconnect

456vv, 2018-05-06:
[image]

The softAP signal is unstable and the connection is often disconnected.

FayeY, 2018-05-07:
"no XXX found" means that the ESP cannot find the target AP to connect. 
You can call wifi_station_disconnect to stop reconnecting. And wifi_station_set_auto_connect(0) to disable the auto connecting.

456vv, 2018-05-07:
The XXX AP signal does exist because the esp8266 could not find the AP. Another problem occurred during this period. The softAP does not work.